deviceplug/btleplug #12: Add iOS Support

qdot, 2020-01-11:
Not sure how much this relates to #2, i.e. how much of CoreBluetooth code can translate between macOS and iOS. Also need to provide documentation on what's required for getting bluetooth access working in iOS apps with rust code (info.plist additions, background perms, etc).

I have very little Cocoa experience, and all my iOS has been through not-quite-native libraries/frameworks (Xamarin/React Native/etc), so lots of open questions.

qdot, 2020-01-21:
Now that we have macOS, this might be done? Need to get an iOS/Swift grownup in to check this.

imalsogreg, 2022-01-07:
@schell can this be closed, given your findings with objective-c?

schell, 2022-01-07:
We have successfully scanned peripherals on iOS using `btleplug`. It seems the only special thing needed in our case (after the expected work of creating a C wrapper/library to FFI into) was the addition of the `key: value` pair `NSBluetoothAlwaysUsageDescription: Our app uses bluetooth to find, connect and transfer data between different devices`. All in all `btleplug` has been great to work with.

qwandor, 2022-01-07:
`NSBluetoothAlwaysUsageDescription` is in the `Info.plist` right?

qdot, 2022-01-07:
We might want to call this closable after we update the README to mention support since it's been confirmed, though it might also be nice to get some basic build instructions in too.

schell, 2022-01-07:
@qwandor yes, exactly. Sorry I was a bit tired last night!

schell, 2022-01-07:
For build instructions it roughly goes: 

* write a rust library (static) that uses `btleplug` and exposes an FFI API to C 
* use `cbindgen` to generate a C header file for that API 
* use `cargo-lipo` to build a universal static lib 
* drag the header file and the library into your Xcode project
* add `NSBluetoothAlwaysUsageDescription` to your `Info.plist` file

qdot, 2022-07-30:
Since we've got iOS working for multiple people now, both via bare library and flutter, calling this done.

closed 2022-07-30 by qdot