deviceplug/btleplug #13: Add WASM/WebBluetooth Support

qdot, 2020-01-11:
Not particularly sure how it'll be useful, but its *the future*, so why not.

Pretty much all of the API redesigns will need to happen before this I'm guessing, as we'll need to figure out how futures will convert and what not. Lots to think about.

qdot, 2020-06-25:
This will basically be a wrapper around wasm_bindings WebSys, which still needs a gecko style WebIDL impl for WebBluetooth written for it.

Guess who's probably doing that. :|

qdot, 2020-09-20:
WebBluetooth IDL support has landed into web-sys, and a WASM implementation is available at https://github.com/buttplugio/buttplug-rs-ffi in the WASM module. This could probably be ported over here at any time (though some web-sys IDL generation fixes are still needed)

qdot, 2020-09-20:
BTW, if someone sees this and wants to try to do it, you will most likely want to talk to me first. Getting the lifetimes right for the WASM/JS execution model is tricky.

arlyxiao, 2023-03-12:
How's this going?
Is this still in progress?

SyedKaz00, 2023-04-03:
^^ Any Progress?

ActuallyHappening, 2023-04-07:
Having WASM support would be awesome!

Since it is not currently supported, I have resorted to using this crate and the web-sys using an even more abstracted API

adiibanez, 2025-03-06:
Hi. I'm currently in the process of integrating btleplug to access ble from elixir via rustler https://hex.pm/packages/rustler_btleplug.
I'm also using some basic javascript web ble in the frontend; in particular for quick demo purposes.

Noticed this issue regarding supporting web bluetooth and was wondering what the point is of accessing ble via webassembly instead of rust or javascript? In a typical scenario what would be the scope and target application to integrate a webassembly btleplug library? wasm runtime on "anything"?

qdot, 2025-03-06:
> Noticed this issue regarding supporting web bluetooth and was wondering what the point is of accessing ble via webassembly instead of rust or javascript? In a typical scenario what would be the scope and target application to integrate a webassembly btleplug library? wasm runtime on "anything"?

Because sometimes you want to write rust to run on the web. For instance, the library I maintain on top of this crate is written in rust, but can compile to WASM and run on the web as well as on various native desktop/mobile platforms. It gives me one codebase for all platforms. A WASM btleplug will still call into the WebAPIs like WebBluetooth (I wrote the webapi bindings for that), but that will be hidden behind our generalized API.