deviceplug/btleplug #8: Add Android Support

qdot, 2020-01-11:
Same idea as blurmac. Fork bluedroid, bring in code, fit to whatever our API surface is.

mvniekerk, 2020-07-23:
Hi - I see this sits on Jan 11. How can I help?

kitlith, 2021-03-26:
may be worth looking into using https://github.com/jni-rs/jni-rs when pulling in blurdroid, it currently appears to use rust -(ffi)-> c -(jni)-> java wrapper -> android bluetooth. I'd think we'd be able to strip out the C layer, at least.

gedgygedgy, 2021-06-04:
I intend to work on this, because I want it for one of my own projects.

I looked into blurdroid, and I see three problems with it:

1. As mentioned above, it has a C middleware layer which can and should be stripped out and replaced with pure Rust.
2. It is asynchronous in nature, due to Android's asynchronous Bluetooth API. This needs to be made synchronous.
3. And of course, it does not match the btleplug API whatsoever.

Changing these three things effectively amounts to a complete rewrite. Blurdroid will certainly serve as a useful reference, but I believe it will be easier to start from scratch.

qdot, 2021-06-04:
@gedgygedgy You'll definitely want to work off the dev branch too, which has an API completely reworked for rust async. It's 100+ commits ahead of master, and we'll hopefully be merging it with master soon, so consider it the place to start.

gedgygedgy, 2021-06-08:
The new async API definitely changes things, and will probably make it easier to port to Android.

One problem I see is that doing anything Java-related in Rust requires a `JavaVM`, which unfortunately can't be obtained from `JNI_CreateJavaVM()` or `JNI_GetCreatedJavaVMs()`. It can only be obtained by implementing `JNI_OnLoad()` in your library or as a parameter to a JNI function. This makes it impossible to implement Java support without either:

1. Adding parameters to `Manager::new()` or
2. Creating a global variable (ick!) which is initialized by `JNI_OnLoad()` (or some init function which gets called by another library), similar to how blurdroid does it.

`Manager::new()` is not currently bound by a trait, which makes it tempting to simply add a `JavaVM` argument to `Manager::new()`, but this would make things slightly more difficult for Android clients, which now have to conditionally (by `#[cfg]`) pass a `JavaVM` around in their Bluetooth code.

Using a global init approach limits the library to a single Java VM (which isn't a problem in Android, since Android only allows one VM per process anyway), requires the client to call the init function if it already implements `JNI_OnLoad()`, and makes Rust angry by creating global variables.

Neither option is very appealing. What are your thoughts? Any other ideas on how to make this work?

gedgygedgy, 2021-06-08:
Personally, I would prefer to add a parameter to `Manager::new()`. The `Manager` will (hopefully) be created early on, at a high level in the program, where it will be easier to conditionally pass a `JavaVM` around.

kitlith, 2021-06-09:
personally, I'm more inclined to go the route of the global variable than I am of adding an argument to the manager, just to reduce the amount of `#[cfg]`s necessary. Expose a function to set the JavaVM for the library, use a OnceCell or similar to store it, call it a day. I will note that said function does not need to be called in JNI_OnLoad (though it is convienent), just sometime before someone calls Manager::new().

it also means that you can have a library A, which uses btleplug and doesn't care about android at all, and application B that runs on android, which uses library A, and application B can just pass the JavaVM to btleplug, still without needing library A to care at all.

gedgygedgy, 2021-06-10:
After seeing how easy `OnceCell` is to use (doesn't even require `unsafe`), I am now thoroughly convinced that a `blteplug::platform::init()` approach is the right way to go.

I've started working on the Android port. You can check out my progress [here](https://github.com/gedgygedgy/btleplug/tree/android).

gedgygedgy, 2021-06-18:
I've created a small crate, [jni-utils-rs](https://github.com/gedgygedgy/jni-utils-rs), to do some extra stuff that jni-rs doesn't do, specifically async and futures. I'm using this as the foundation of my Android port.

gedgygedgy, 2021-06-18:
What is btleplug's expectation for attempting to run two `async` commands at the same time? For example, one thread calls `Peripheral::connect()`, and then another thread calls `Peripheral::disconnect()` before `Peripheral::connect()` is finished. Is btleplug supposed to handle this, or is the user just supposed to not do this? Android's Bluetooth system doesn't seem to handle simultaneous commands very well, so I'm wondering if I need to set up some sort of command queue.

qdot, 2021-06-18:
> Upon thinking about this some more, it doesn't seem very Rustacean to provide an API that can fail so easily. I will go forward with a command queue to make sure only one command gets executed at a time.

Well if it makes you feel any better, I was gonna say command queue too. It's what I usually do in this case. :)

gedgygedgy, 2021-06-24:
I have a working `Peripheral::connect()` and `Peripheral::disconnect()` implementation. It's a bit of a proof-of-concept and needs to be cleaned up with a proper command queue, but it works.

qdot, 2021-06-24:
> I have a working `Peripheral::connect()` and `Peripheral::disconnect()` implementation. It's a bit of a proof-of-concept and needs to be cleaned up with a proper command queue, but it works.

WOOOOOOOOOOOOOOOOOOOO 👍🏻 💯 🥇 🚀

gedgygedgy, 2021-06-25:
I've added a command queue to ensure commands execute one at a time. `Peripheral::discover_services()`, `Peripheral::read()`, and `Peripheral::write()` are all working. The rest of the functionality is on its way.

gedgygedgy, 2021-07-02:
At this point I think I've implemented all of the functionality for Android. Scanning, `CentralEvent`s, reading/writing characteristics, peripheral properties, and characteristic notifications are all working.

A few caveats:

* This is somewhat dependent on my API changes in #166.
* This is also dependent on [jni-utils-rs](https://github.com/gedgygedgy/jni-utils-rs), which has not yet stabilized, and which I have not yet published to crates.io.
* Android apps that use this functionality must declare a runtime-only dependency on the `.aar` file generated by `src/droidplug/java` (which itself also depends on the Java code from jni-utils-rs.)
* This code interacts heavily with the JVM. As such, any thread that interacts with btleplug must be attached to the JVM. If the thread is created natively (rather than by a `java.lang.Thread`), it must call `JavaVM::attach_current_thread()`, and it must also set the thread's classloader. While this isn't a dealbreaker for systems like tokio, it does make them more difficult to use.
* For some reason, not all events that get sent by `AdapterManager::emit()` seem to go through, at least on my setup. (Strangely, adding print statements between the `emit()` calls seems to help somewhat. Perhaps there's a race condition here?)
* Some apps won't want to use `Adapter::start_scan()` and `Adapter::stop_scan()`, because this requires location permissions on Android. To allow such apps to use the Companion Device Manager instead, I've provided an `Adapter::report_scan_result()` function, which takes an `android.bluetooth.le.ScanResult` and applies it as if it had been discovered by its own scan. In addition, clients can use `Adapter::add_peripheral()` to skip the scanning step if they already know the MAC address (probably retrieved from the Companion Device Manager.)
* I have not written *any* documentation yet. That is going to be my next step.

kitlith, 2021-07-02:
> This code interacts heavily with the JVM. As such, any thread that interacts with btleplug must be attached to the JVM. If the thread is created natively (rather than by a `java.lang.Thread`), it [...]

it seems like tokio has a couple of things that might help with this and not be too much of a hassle (in the case where you attach every thread to the jvm at least): https://docs.rs/tokio/1.7.1/tokio/runtime/struct.Builder.html#method.on_thread_start and https://docs.rs/tokio/1.7.1/tokio/runtime/struct.Builder.html#method.on_thread_stop

schell, 2022-01-01:
What's the status on this? Did any of these changes make it into 0.9?

qdot, 2022-01-01:
This is currently on hold. The original author went quite in mid-August 2021, and I haven't been able to get ahold of them since.

stevekuznetsov, 2022-03-26:
@qdot hi! Is there any chance we might take over the work from the original author? It has been quite a bit since they last showed interest, right?

qdot, 2022-04-16:
@stevekuznetsov I actually just started working on this exact thing! I've at least gotten their branch to build, but we'll need to bring it up to the 0.8 API, which I don't think should be *too* much work.

They were doing this work in order to work with my other library, so my goal right now is to get that full chain built and working on a phone from the state it was in last year, then start updating the pieces from btleplug up. I'll let you know how this turns out.

qdot, 2022-04-18:
Good news! I've managed to get @gedgygedgy's full chain from btleplug up through buttplug running on my Pixel 3. This is going to require a TON of documentation, as said earlier, btleplug android (aka droidplug) expects to use its own async executor or thread registration, in order to keep the environment/thread contexts straight. Buttplug has an executor abstraction mechanism to handle this currently, but I'm... not real sure how to convey this requirement in relation to btleplug for normal users.

The focus right now will be getting the old repo from btleplug 0.7 to btleplug 0.9, then I'll take a look at the executor context stuff for tokio.

@stevekuznetsov @schell

stevekuznetsov, 2022-04-18:
Wow that's super exciting! Please let me know when you get to a good spot if there's something I could help out with :)

schell, 2022-04-21:
@qdot that's great news! What branch is this work happening on? I'd like to follow along.

qdot, 2022-04-21:
The work on bringing up android support to the 0.9.2 API is done and working, now I'm on to tokio runtime support, which is mostly there but hitting a some JNI bugs. Hopefully will get that ironed out in the next few days. I need to do some cleanup because I've been moving kinda fast and loose on this, but should have a branch up tomorrow.

@schell @stevekuznetsov

qdot, 2022-04-23:
Ok, the android branch is now up as `android-update` on this repo. I'll open a PR on it also, just so others will know, but I have a feeling it may be a while before it comes in.

Unless you are well versed in JNI and poking at gradle by hand, I wouldn't recommend playing with this quite yet.

In terms of how to build it:

- You'll need to clone gedgygedgy/jni-rs, gedgegedgy/jni-utils-rs, and gedgygedgy/android-utils-rs and have them next to this repo in the file system. I currently just set up local path dependencies in cargo between all of them.
- You'll write/build your rust code as normal, I recommend using cargo ndk.
- You'll need to local path links from your app gradle to this library as well as jni-rs probably. You'll also need to call init on all the libraries on JNI_OnLoad, it looks something like

```rust
#[no_mangle]
pub extern "C" fn JNI_OnLoad(vm: JavaVM, _res: *const c_void) -> jint {
    let env = vm.get_env().unwrap();
    android_utils::init(&env).unwrap();
    jni_utils::init(&env).unwrap();
    btleplug::platform::init(&env).unwrap();
    jni::JNIVersion::V6.into()
}
```

- For the java, the gradle files are a total mess. I'd recommend setting local path dependencies there too, I have them set to push to a local maven instance right now but that's kinda silly.
- Currently you'll have to use the gedgygedgy/android-utils-rs runtime. Here's the implementation of that under my other library's async abstraction layer: https://github.com/gedgygedgy/buttplug/blob/android/buttplug/src/util/async_manager/android.rs

So yeah, using this at the moment is a *significant* amount of work. 

I'm still stuck on tokio support right now. The find_class() calls to get the btleplug Peripheral java class fail when called on non-main threads, even if the thread is attached to the VM and has its classloader set to the main thread classloader. No idea what's up. I've got some vague solutions around caching ClassLoader calls during initialization that *might* do the trick but it's not a great solution.

qdot, 2022-04-23:
Ok, well, went ahead and implemented the ClassLoader cache and it makes the tokio multithreaded runtime work. That said, it's ugly and requires changes back through jni_utils. Would really like to figure out why env.find_class() isn't working but we have a backup solution if we absolutely need it.

qdot, 2022-04-24:
Finally did more cleanup and got the tokio code up. This requires:

- https://github.com/deviceplug/jni-utils-rs, classcache branch
- https://github.com/deviceplug/btleplug, android-tokio branch

**THE android-utils-rs REPO IS NO LONGER NEEDED WHEN USING TOKIO AND THESE NEW BRANCHES**

So you should be able to:

- Either build java directories or add them to your app gradle as local path dependencies
- Set the btleplug jni-utils-rs dep to your local checkout on the classcache branch (I think I have the cargo file already doing this and assuming the repo directories are next to each other)
- Build btleplug using `cargo ndk`
- Copy those output files to the proper native lib architecture directory in your app.

Lemme know if you actually start trying this and either automate those steps in gradle (which I barely understand how to use) or need me to list out how I'm running things.

schell, 2022-04-25:
This is great progress @qdot ! Thank you so much!

trobanga, 2022-05-19:
Hi, I am playing with this and actually managed to build a flutter app that starts without error. I'm trying the event_driven_discovery example inside a 
```rust
tokio::runtime::Builder::new_multi_thread()
        .enable_all()
        .build()
        .unwrap()
        .block_on(async {
```
block. Unfortunately, line `central.start_scan(ScanFilter::default()).await.unwrap();` fails with JniCall(ThreadDetached). Do you have any ideas what's happening there?

qdot, 2022-05-19:
@trobanga 

Oh wow, that's awesome that you've got it working through flutter! Curious about your FFI setup for that. :)

If you're seeing ThreadDetached, that most likely means the required setup methods haven't been run. Unfortunately I don't have great (or really any) documentation for this yet, but I'll point you to where I implemented this and gesture wildly in a hopefully helpful way.
 
First off, you'll need to have a JNI_OnLoad method that calls all of the proper init methods across the required libraries:

https://github.com/buttplugio/buttplug-rs-ffi/blob/52b8ff0e49dc4b0ce34f254a790ae4209ff0c422/ffi/src/export.rs#L29

JNI_OnLoad is called by JNI when loading the library, so it should just be called automatically when library is loaded into the JVM.

Then, you'll need to set up your Tokio runtime to attach threads to the JNI as they're created:

https://github.com/buttplugio/buttplug-rs-ffi/blob/52b8ff0e49dc4b0ce34f254a790ae4209ff0c422/ffi/src/export.rs#L56

After that, things *should* work. Or hopefully you'll at least get a different error.

trobanga, 2022-05-19:
No, it's still the same error. I'll take a closer look tomorrow.

qdot, 2022-05-19:
Well, ThreadDetached means that whatever thread the btleplug command is running from isn't attached to the JVM. That needs to happen on thread bringup, otherwise there's no way for the JNI portions of the library to reach the Android stuff. Not real sure how this would work in flutter's library loading and execution context tho.

trobanga, 2022-05-20:
It works! In the end I just had to run `vm.attach_current_thread()` in the `block_on` closure too.
I've created an example [flutter_btleplug](https://github.com/trobanga/btleplugtest). If you go to example/ and type `flutter run` it should immediately work an throw the scanned devices on the console.

trobanga, 2022-05-20:
Depending on your Android device the app might need permission to location.

qdot, 2022-07-24:
Back at this, now trying build my library chain into my own flutter app. Assuming all of that doesn't end in a giant ball of fire, I'll probably just end up maintaining jni-utils-rs so we can bring the android branches into btleplug mainline finally.

qdot, 2022-07-28:
Ok! Got blteplug android up and working in my own flutter project. Testing iOS next.

@trobanga Not sure if you're still working on your app with btleplug, but if you want to avoid having to sprinkle attach_current_thread() everywhere, you can put your AttachGuard in thread local storage, and just free it when the thread terminates. Probably faster to do that too. I've got a gist with my modified version of your android setup code here:

https://gist.github.com/qdot/210f31439888927fc23b14d2e3d40503

Now we're just down to figuring out what and how to get everything landed.

qdot, 2022-07-30:
Ok, Android is now a supported platform in v0.10. Thanks to everyone who helped out on this!

closed 2022-07-30 by qdot

qdot, 2022-10-21:
@trobanga Just curious, have you faced any issues with dead code removal when compiling flutter apps for release with btleplug? My flutter app works fine in debug, but crashes in release on android, in a way similar to what I was seeing when I'd had the wrong dependency for the droidplug.aar file that means it wasn't getting included. I'm wondering if I need to add some sort of explicit call into droidplug to keep it live or something, but that's just a guess at the moment.

trobanga, 2022-10-22:
I just checked and in release mode my btleplugtest crashes immediately. Unfortunately I'm no Android expert and I don't have time to dig in at the moment.

qdot, 2022-10-22:
@trobanga Ok cool, thanks for checking that! I just filed #272, and will be working on this myself. Pretty sure I just need to add some sort of explicit access to the aar to keep the file alive.

denisidoro, 2023-01-07:
Hey! Great work on adding support for Android!

I think I already know the answer but just confirming: is it possible to build a working CLI for Android? 

No UI, no flutter, no maven. Just a single binary (built with cross, for example) that I can call from Termux.

Thanks!

qdot, 2023-01-07:
@denisidoro I... honestly have no idea. I'm not sure what kinda environment termux executes in, so I'm not sure if the JNI the process normally requires work would?

loikki, 2024-02-13:
Hey,

@qdot Thanks a lot for your library. You did an amazing work with android. Unfortunately, the documentation is a bit shallow and I lost a bit of time figuring out how to do it. So let me drop a few comments to help people without a good understanding of android / java.

I managed to get the bluetooth working with tauri. You can check my code [here](https://gitlab.com/loikki/open-biking/-/tree/9b260320b5521a2f7b239279ec796dcc4b2fef2a). This is the first commit after I finally managed to get it working, so it is extremely dirty. I am expecting the code to be cleaner in a near future. There is a link to my discord in the readme if anyone need to reach me regarding my git repository.

The few key points to get it working:
 - Add permissions to your application (I might be asking too much here): [AndroidManifest.xml](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/android-files/AndroidManifest.xml)
 - Ask for permissions in your application: [MainActivity.kt](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/android-files/MainActivity.kt)
 - Initialize btleplug in "java": [JNI_Onload](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/src/mobile.rs#L136)
 - Add the java code from both jni-utils-rs and btleplug to your android application: [Makefile](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/Makefile#L81)
 - Start a "java" thread for managing the bluetooth: [create_runtime](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/src/mobile.rs#L57)
 - Only ever access the bluetooth interface from within the thread: [scan](https://gitlab.com/loikki/open-biking/-/blob/9b260320b5521a2f7b239279ec796dcc4b2fef2a/src/bike.rs#L37)
 
 @NuLL3rr0r In your update, it looks like you forgot to call start_scan in the "java" thread.

qdot, 2024-02-13:
Yeah, I'm holding off on writing full documentation about compiling for mobile until I know the system actually *works* everywhere. I'm shipping this in an app with ~10k installs at the moment and we're still seeing a steady stream of very weird, untraceable crashes. Not only that, sometimes the app won't crash but also just won't connect to anything (a big problem on Samsung phones, for some reason).

In terms of the instructions you provided (looks like you grabbed the init code from the rust bridge in intiface/intiface-central)? , do note that this method will work for keeping the app focused, but scanning and continuing connections while foregrounding may require extra permissions.

qdot, 2024-02-13:
And here's what your permissions should ideally look like, as you'll want to restrict them based on the Android version your app is running on: https://github.com/intiface/intiface-central/blob/main/android/app/src/main/AndroidManifest.xml

loikki, 2024-02-14:
Thanks for the hints! I will update everything.

It is mostly based on the following codes but yeah I looked a bit around at your code:
 - https://github.com/trobanga/flutter_btleplug
 - https://gist.github.com/qdot/210f31439888927fc23b14d2e3d40503

qdot, 2024-02-14:
Ah, ok, yeah, here's the current version of that gist, which I adapted off of trobanga's original shim layer: https://github.com/intiface/intiface-central/blob/main/intiface-engine-flutter-bridge/src/mobile_init/setup/android.rs

trobanga, 2024-02-15:
@qdot interesting to see that you are using flutter_rust_bridge, too. Have a look at [frb2](https://github.com/trobanga/flutter_btleplug/tree/frb2) if you are interested in an adaption to flutter_rust_bridge v2.

Gibbz, 2024-07-26:
Has there been any follow-up on a guide for getting this working?
I'm about to start the process myself. And the info seems fragmented... See how I go!

lnxdxtf, 2024-07-29:
Hi everyone,

I'm creating a Tauri plugin to use thermal printers. The plugin handles all permissions (ask and check) to use BLE. The crate btleplug is wrapped in a crate called [eco_print](https://github.com/lnxdxtf/eco_print). It works on desktops, but I'm having trouble making it work on mobile (Android).

I'm trying to find a way to build the crate for use in my plugin, [tauri-plugin-escpos](https://github.com/lnxdxtf/tauri-plugin-escpos). However, I'm not sure about the exact steps needed to build the crate for Android. Any guidance on this would be greatly appreciated.

Thank you!

loikki, 2024-07-29:
Hi @lnxdxtf 

I would recommend taking a look at my repo. I have a tauri app with bluetooth for an indoor bike: https://gitlab.com/loikki/open-biking
The modifications to the android files are all located in `android-files` and the commands for building / running are in my Makefile. If you have any question, feel free to open an issue in my repo

qdot, 2024-07-29:
For anyone that gets btleplug running on android, please let me know! I'll try to remember to add project listings to the readme so other people can use them as examples. Posting project links here is fine.

TheCataliasTNT2k, 2024-11-12:
I am trying to get this to work as as CLI app built using cross, just like @denisidoro.
Any instructions, how to do this?

zhenzhongli, 2024-12-06:
> Hey,
>
> @qdot Thanks a lot for your library. You did an amazing work with android. Unfortunately, the documentation is a bit shallow and I lost a bit of time figuring out how to do it. So let me drop [...]

> Hey,
>
> @qdot Thanks a lot for your library. You did an amazing work with android. Unfortunately, the documentation is a bit shallow and I lost a bit of time figuring out how to do it. So let me drop [...]

Hi, I followed your approach, but in mobile.rs, the line btleplug::platform::init(&env).unwrap(); results in an error saying that the init function cannot be found. Can you explain why?  Thank you!

zhenzhongli, 2024-12-06:
> Hey,
>
> @qdot Thanks a lot for your library. You did an amazing work with android. Unfortunately, the documentation is a bit shallow and I lost a bit of time figuring out how to do it. So let me drop [...]

Hi, I followed your approach, but in mobile.rs, the line btleplug::platform::init(&env).unwrap(); results in an error saying that the init function cannot be found. Can you explain why? Thank you!

loikki, 2024-12-10:
Can you send the logs please? Without them it is hard to understand what is going on. I am a bit rusty on the bluetooth side, so I might be asking the wrong questions.

Is it during compilation or running?

If during compilation:
- Are you using the same version of the library than me?

else:
- Is the java environment well set when you call the function? 
- On the right thread? 
- Do you have the required permission for bluetooth?

zhenzhongli, 2024-12-11:
> Can you send the logs please? Without them it is hard to understand what is going on. I am a bit rusty on the bluetooth side, so I might be asking the wrong questions.
>
> Is it during compilation or [...]

I am currently working on a project using Tauri 2 and this library. Sorry, I can't provide log information as I have no experience with Android or Java. The main issue is that I don't know what tasks need to be done, what specific modifications are required, or the steps to follow. Adapting this library to Android is indeed quite challenging for me.

XavierIsabel, 2024-12-23:
Hi @loikki @qdot , I am building a flutter app connected to a Rust backend using btleplug, similar to your Tauri app you gave as an example. I am now able to scan, but rust panics when I try to connect to the ble device I want to connect to. This is the error:
```
I/flutter (29654): panicked at C:\Users\xavie\.cargo\registry\src\index.crates.io-6f17d22bba15001f\jni-utils-0.1.1\rust\future.rs:42:98:
I/flutter (29654): called `Option::unwrap()` on a `None` value
```
Would you know why? A similar problem occured for @lnxdxtf in https://github.com/deviceplug/btleplug/issues/395 but the answer was never given. This is my rust code to interface ble with android. Excuses in advance for my rust, I am learning at the same time. I can give more info if needed.

```
#[cfg(target_os = "android")]
use std::sync::Once;
#[cfg(target_os = "android")]
use once_cell::sync::OnceCell;

#[cfg(target_os = "android")]
pub static JAVA_VM: OnceCell<JavaVM> = OnceCell::new();

#[cfg(target_os = "android")]
use jni::{JNIEnv, JavaVM};
#[cfg(target_os = "android")]
use jni::objects::{JClass, GlobalRef};
#[cfg(target_os = "android")]
use jni::errors::Error;

#[cfg(target_os = "android")]
static INIT: Once = Once::new();

#[cfg(target_os = "android")]
#[no_mangle]
pub extern "system" fn Java_com_example_sifi_1gui_RustLibBLE_initializeJavaVM(env: JNIEnv, _: JClass) {
    let java_vm = env.get_java_vm().expect("Failed to get JavaVM");

    JAVA_VM.set(java_vm);

    INIT.call_once(|| {
        btleplug::platform::init(&env);
    });
}

#[cfg(target_os = "android")]
fn get_java_vm() -> &'static JavaVM {
    JAVA_VM.get().expect("JavaVM is not initialized")
}

#[cfg(target_os = "android")]
pub fn attach_thread_to_jvm() -> Result<JNIEnv<'static>, Error> {
    let java_vm = get_java_vm();
    java_vm.attach_current_thread_as_daemon()
}

#[cfg(target_os = "android")]
pub fn detach_thread_from_jvm() {
    let java_vm = get_java_vm();
    java_vm.detach_current_thread();
}

#[cfg(target_os = "android")]
pub fn setup_class_loader(env: &JNIEnv) -> Result<GlobalRef, Error> {
    let thread = env
        .call_static_method(
            "java/lang/Thread",
            "currentThread",
            "()Ljava/lang/Thread;",
            &[],
        )?
        .l()?;
    let class_loader = env
        .call_method(
            thread,
            "getContextClassLoader",
            "()Ljava/lang/ClassLoader;",
            &[],
        )?
        .l()?;

    Ok(env.new_global_ref(class_loader)?)
}
```

qdot, 2024-12-23:
I'm not sure why people keep replying to a closed issue instead of filing new bugs, but I'm gonna go ahead and lock this issue. Please file new issues for things that are, you know, new issues.