deviceplug/btleplug #9: Move bluez's dbus to using dbus-tokio

qdot, 2020-01-11:
Would be nice to have async all the way down for linux via dbus-tokio's (https://github.com/diwic/dbus-rs) support for futures 0.3/tokio 0.2. Just gotta figure if we need a pure tokio executor or if any executor would work now.

blandger, 2020-07-26:
@qdot 
Do you think about attempt to switch server's code to async flow access ? So clients can make async access too?

qdot, 2020-07-26:
@blandger Ok, well, closing this specific bug because this library doesn't even use dbus, it's using direct socket access to bluetoothd. Didn't realize this bug was still live heh.

That said, yes, the plan is to do a complete async overhaul for btleplug v1.0. The changes I'm making right now (moving toward v0.5) are to try and get things to a point where the library is usable while that happens, because that's probably a multi-month project.

closed 2020-07-26 by qdot