deviceplug/btleplug #19: central.peripherals() panics on linux

seymour-bootay, 2020-01-28:
I am running a modified version of the light.rs example, I removed everything after line 45 and changed "LEDBLUE" to the local name of a device broadcasting near me, to attempt to see advertising devices from linux.  I am getting the following panic on the central.peripherals() call:

thread 'main' panicked at called Option::unwrap() on a 'None' value, src/libcore/option.rs:378:21
stack backtrace:
   0: backtrace::backtrace::libunwind::trace
             at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunwind.rs:88

   1: backtrace::backtrace::trace_unsynchronized
             at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs:66

   2: std::sys_common::backtrace::_print_fmt
             at src/libstd/sys_common/backtrace.rs:77

   3: <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt
             at src/libstd/sys_common/backtrace.rs:61

   4: core::fmt::write
             at src/libcore/fmt/mod.rs:1028

   5: std::io::Write::write_fmt
             at src/libstd/io/mod.rs:1412

   6: std::sys_common::backtrace::_print
             at src/libstd/sys_common/backtrace.rs:65

   7: std::sys_common::backtrace::print
             at src/libstd/sys_common/backtrace.rs:50

   8: std::panicking::default_hook::{{closure}}
             at src/libstd/panicking.rs:188

   9: std::panicking::default_hook
             at src/libstd/panicking.rs:205

  10: std::panicking::rust_panic_with_hook
             at src/libstd/panicking.rs:464

  11: std::panicking::continue_panic_fmt
             at src/libstd/panicking.rs:373

  12: rust_begin_unwind
             at src/libstd/panicking.rs:302

  13: core::panicking::panic_fmt
             at src/libcore/panicking.rs:139

  14: core::panicking::panic
             at src/libcore/panicking.rs:70

  15: core::option::Option<T>::unwrap
             at /rustc/73528e339aae0f17a15ffa49a8ac608f50c6cf14/src/libcore/macros.rs:41

  16: ml::main
             at examples/ml.rs:45

  17: std::rt::lang_start::{{closure}}
             at /rustc/73528e339aae0f17a15ffa49a8ac608f50c6cf14/src/libstd/rt.rs:61

  18: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:48

  19: std::panicking::try::do_call
             at src/libstd/panicking.rs:287

  20: __rust_maybe_catch_panic
             at src/libpanic_unwind/lib.rs:78

  21: std::panicking::try
             at src/libstd/panicking.rs:265

  22: std::panic::catch_unwind
             at src/libstd/panic.rs:396

  23: std::rt::lang_start_internal
             at src/libstd/rt.rs:47

  24: std::rt::lang_start
             at /rustc/73528e339aae0f17a15ffa49a8ac608f50c6cf14/src/libstd/rt.rs:61

  25: main

  26: __libc_start_main

  27: _start

qdot, 2020-01-28:
First off, can you run this in nightly? It'll hand back a more useful backtrace, since there's now code in there to resolve panics outside of the std origination point.

Secondly, the example itself (which I just took wholesale from Rumble and haven't changed at all) doesn't do any actual checking for device validity. It's just assuming that your light will be found in 2 seconds, which isn't necessarily true. I might suggest changing this to use the on_event handler instead of a random spinwait. You can also watch the logs by adding an env_logger call and then outputting log messages at the `Debug` or `Trace` level, the library is super spammy during scanning so you'll be able to see there if/when your device comes up in the scan.

seymour-bootay, 2020-01-29:
Thanks I am new to rust, adding the env_logger call was extremely helpful.
The Peripherals returned from central.peripherals() only have address, address_type, discovery_count, and has_scan_response(this is always false) set. local_name, tx power, and manufacturer_data do not have values.
I set a breakpoint in the `handle_device_message(&self, message: &hci::Message)` method on line 149 of bluez/adapter/peripheral.rs, where I believe it should be getting the local_name but it is never hit. I do however see tx_power_level (line 152) and manufacturer_data (line 155) being set, but the data is not returned when I call central.peripherals() to get the discovered devices. Are you able to get the local_name or manufacturer_data on linux? I am currently using the master branch, should I be using dev? I will try to debug further tonight.

qdot, 2020-01-30:
Ok, we ran into this problem on windows/macOS too, where we'd get advertisements with varying amounts of data through for some reason. I think there may be something weird about how advertisements get built, but since bluez was already built as part of the original rumble code and it was working ok for me so I didn't think much of it. I'll take a look too.

seymour-bootay, 2020-01-30:
My hunch is that it is an issue with the parsing in the le_advertising_data function located in hci.rs. I think try_parse(i, map!(take!(len), |b| vec![LocalName(String::from_utf8_lossy(b).into_owned())]))
may be failing.  All the values seem to be correct at this point but it looks like the try_parse is failing.  I will have to read up on what try_parse is doing, I am not entirely sure.

seymour-bootay, 2020-02-03:
I am seeing what I would expect from the ADV_IND (Advertising Flags, Service UUIDs, Slave Connection Interval, and Tx Power Level) but I am not seeing the SCAN_RSP (scan response) data.

seymour-bootay, 2020-02-03:
I am now getting data from the scan responses, not sure what fixed it.  I added central.active(true); to my example code (which shouldn't have mattered because I think it defaults to active scanning).  I will see if I can reproduce the error state again and report back.

blandger, 2020-02-18:
@seymour-bootay 
I wrote some code for Linux and found out following approach to skip 'incorrectly found' BLE peripheral. 
`peripheral.properties().local_name..is_some()`

```
for peripheral in connected_adapter.peripherals().iter() {
     if peripheral.properties().local_name.is_some() && !peripheral.is_connected() {
              peripheral.connect().expect("Can't connect to peripheral...");
              let chars = peripheral.discover_characteristics();
              if peripheral.is_connected() {
                    for chars_vector in chars.into_iter() {
                          for char_item in chars_vector.iter() {
...........
```

seymour-bootay, 2020-03-20:
Sorry for the delayed response, thanks blandger your code was helpful.  I am closing as this is no longer an issue for me.

closed 2020-03-20 by seymour-bootay

mchodzikiewicz, 2020-04-04:
Hey, should this be considered closed? #20 which was closed as duplicate, but it still exists ie. using btleplug
```
let devices: Vec<Option<String>> = central.peripherals().into_iter()
    .map(|dev| dev.properties().local_name).collect();
    println!("Available devices:\n{:?}", devices);
```
 consistently gives me something like this
```
Available devices:
[None, None, None, None]
```
where rumble gives nice
```
Available devices:
[None, None, Some("Tilt")]
```