deviceplug/btleplug #25: Cannot reconnect to Peripheral after Disconnect

IamfromSpace, 2020-03-05:
My code is setup to connect to a peripheral, subscribe to it, output its notifications to the console.  It then attaches a subscription to the central, which logs all events and attempts to reconnect to any address that appears in a DeviceDisconnected event.

I then execute the program, wait for notifications, physically shield the peripheral wait for the disconnection, then immediately unshield the device.

**Expected behavior**
The device should reconnect.

**Actual behavior**
There is a 20s timeout error.

**Additional context**
I have also tried adding 30s delays, trying to disconnect from the device before connecting, and trying to start and then stop scanning before re-connecting.  None of these things seem to work.  This is a Polar H10 HRM, I have a few other BLE devices I will also try.

qdot, 2020-03-05:
What OS are you on?

IamfromSpace, 2020-03-05:
Good question!  Sorry I left that out.  This is on Linux.

Also I tested this on a Kickr Core bike trainer, which can just be unplugged and plugged back in.  The effect is the same.

However, this has a blue LED with three states: blinking is not connected, fast blink is connecting, and solid is connected.  What’s interesting is that on the call to connect, the light goes solid!  So it appears that the device believes it’s connected, but the central does not.

IamfromSpace, 2020-03-05:
Also just tried this out on my Raspberry Pi Zero W and got the exact same result.

So this is replicated on:
Rasbian 2020-02-05-buster-lite + bluez 5.50
Ubuntu 16.04 + bluez 5.41

IamfromSpace, 2020-03-15:
So digging into this, it appears that the issue appears to be _half_ my fault, but this is very un-intuitive, and likely needs a fix.  The issue is that when I was listening to disconnects, I was immediately trying to connect back to that same device that was disconnected in a blocking fashion.  What's not so obvious is that:
  a) The event callback is on the same thread as the socket reading thread.
  b) A connection cannot succeed until it receives a connection success event from the socket.

Essentially, I caused the deadlock with my, "if you see a disconnect, reconnect" behavior.  I can theoretically fix this by handling this in a non-blocking way.

However, rather than put this requirement on a user, it likely makes more sense to call event handlers in a separate thread, so that no action by the user blocks the socket.

qdot, 2020-07-27:
As of v0.5.0, events are now sent on a channel, so this shouldn't be a problem anymore. Closing.

closed 2020-07-27 by qdot