deviceplug/btleplug #26: Disconnected Peripheral ends Notifications for all other Peripherals

IamfromSpace, 2020-03-06:
To create this bug I connected to two devices, subscribed to a single characteristic for each, then simply logged their notifications to the console.  Then one (either) of the devices is either powered down or shielded from transmitting data until there is a disconnection detected.

**Expected behavior**
Notifications are still received for the other device.

**Actual behavior**
All notifications end.

**Additional context**
This seems like it may be linked to #25.  In debugging that issue further I discovered this and it seemed different enough in symptom that it was worth capturing separately.

Found on Ubuntu 16.04 + bluez 5.41

qdot, 2020-03-06:
Good catch, thanks! It may be a bit before I can start addressing these, but I'll let you know once I've got time to look.

IamfromSpace, 2020-03-15:
Confirmed that this is _unrelated_ to #25.  That issue is caused by event handlers blocking the socket, but this occurs even if the user doesn't block the socket.

I have some other leads based on troubleshooting #25, but nothing solid yet.

IamfromSpace, 2020-03-15:
I was a bit rushed, and looked like my double-check wasn't thorough enough!  This does appear to be caused by #25 after all, so I'm going to close this.

closed 2020-03-15 by IamfromSpace