deviceplug/btleplug #40: Add an example for writing own Central::on_event handler

vietlq, 2020-04-21:
**Add an example for writing own Central::on_event handler**

Hi, I'm new to Rust and don't quite understand how to write an instance of `EventHandler` that can be passed to Central::on_event. Any quick example would be appreciated. I can update README.md via PR once you show me a basic example by replying here. Thank you.

```
pub type EventHandler = Box<dyn Fn(CentralEvent) + Send>;

/// Central is the "client" of BLE. It's able to scan for and establish connections to peripherals.
pub trait Central<P: Peripheral>: Send + Sync + Clone {
    /// Registers a function that will receive notifications when events occur for this Central
    /// module. See [`Event`](enum.CentralEvent.html) for the full set of events. Note that the
    /// handler will be called in a common thread, so it should not block.
    fn on_event(&self, handler: EventHandler);
```

Moret84, 2020-04-21:
Hi, 

If it helps, I have implemented scanning with callback in one of my project: [rs-mijiabt](https://github.com/Moret84/rs-mijiabt/blob/master/src/btleplug_ble_repo.rs) 

(I am quite new to rust too)

qdot, 2020-04-22:
The way the library currently handles events is not really what I'd call idiomatic rust (I didn't write it, it was adapted from one of the 3 libraries this library pulled from just so we could get cross platform bluetooth ASAP). Passing in closure callbacks is a nightmare for lifetime handling. This should use a stream that you can manage whenever you want, and that's the plan for the near-to-medium future.

For my library right now, I convert to a streaming method by using channels.

https://github.com/buttplugio/buttplug-rs/blob/master/buttplug/src/server/comm_managers/btleplug/btleplug_internal.rs#L62

Basically I just create a channel, and throw the sender into the event handler. Then I can deal with the events whenever I want by either waiting on/racing the receiver.

qdot, 2020-04-22:
Oh yeah, and leaving this open, as I'll see about adding an example for showing how that works so you don't have to go digging through my very odd libraries. :)

vietlq, 2020-04-22:
Thank you @Moret84 and @qdot for your kind responses.

qdot, 2020-07-25:
This landed in April, closing.

closed 2020-07-25 by qdot