deviceplug/btleplug #41: Support print characteristics on macOS

vietlq, 2020-04-21:
**Feature Description**

I took the example from README and adapted it:

```
extern crate btleplug;
extern crate rand;

use btleplug::api::{Central, CentralEvent, EventHandler, Peripheral, UUID};
#[cfg(target_os = "linux")]
use btleplug::bluez::{adapter::ConnectedAdapter, manager::Manager};
#[cfg(target_os = "macos")]
use btleplug::corebluetooth::{adapter::Adapter, manager::Manager};
#[cfg(target_os = "windows")]
use btleplug::winrtble::{adapter::Adapter, manager::Manager};
use rand::{thread_rng, Rng};
use std::thread;
use std::time::Duration;

// adapter retrieval works differently depending on your platform right now.
// API needs to be aligned.

#[cfg(any(target_os = "windows", target_os = "macos"))]
fn get_central(manager: &Manager) -> Adapter {
    let adapters = manager.adapters().unwrap();
    adapters.into_iter().nth(0).unwrap()
}

#[cfg(target_os = "linux")]
fn get_central(manager: &Manager) -> ConnectedAdapter {
    let adapters = manager.adapters().unwrap();
    let adapter = adapters.into_iter().nth(0).unwrap();
    adapter.connect().unwrap()
}

pub fn main() {
    let manager = Manager::new().unwrap();

    // get the first bluetooth adapter
    //
    // connect to the adapter
    let central = get_central(&manager);

    // start scanning for devices
    central.start_scan().unwrap();
    // instead of waiting, you can use central.on_event to be notified of
    // new devices

    let mut count = 0;
    while count < 30 {
        println!("Count = {}", count);
        thread::sleep(Duration::from_secs(1));
        count += 1;
    }

    for p in central.peripherals() {
        match p.discover_characteristics() {
            Ok(v) => println!("Got characteristics: {:?}", v),
            Err(e) => println!("Got errors when querying characteristics: {:?}", e),
        };
        println!("p.properties = {:?}", p.properties());
        println!("p.characteristics = {:?}", p.characteristics());
    }
}
```

There are no characteristics printed on macOS. I understand that this is WIP and with some guidance I could help with adding this feature :) Thanks.

NZSmartie, 2021-03-02:
Hey @vietlq, are you still having this problem? did the code compile for you, and run, but printed nothing for the `println!("p.characteristics = {:?}", p.characteristics());` line? 

The `PeripheralProperties` has a `#[derrive(Debug)]` which would indicate that you should be seeing a printout of the properties for the `PeripheralProperties` struct.
https://github.com/deviceplug/btleplug/blob/c4432d35f858573a4fdcb7dcad4a96fbdc03748d/src/api/mod.rs#L220-L244
As this i defined in `btleplug::api` and not specifically in `btleplug::corebluetooth` part of this library, I would assume others would also be having this problem if that was the case.

qdot, 2021-07-31:
This should work in 0.8 now.

closed 2021-07-31 by qdot