deviceplug/btleplug #43: macOS Connect fails with InvalidLength

mitsuhiko, 2020-05-25:
**Describe the bug**
I'm trying to connect to a peripheral (not sure which one, i just tried to connect to all my peripherals discovered) and ran into a panic:

```
[src/main.rs:63] peripheral.properties() = PeripheralProperties {
    address: 63:31:30:61:31:62,
    address_type: Random,
    local_name: Some(
        "nil",
    ),
    tx_power_level: None,
    manufacturer_data: None,
    discovery_count: 1,
    has_scan_response: true,
}
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error(Parser(InvalidLength { expected: Any([36, 32]), found: 4 }))', /Users/mitsuhiko/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.4.2/src/corebluetooth/central_delegate.rs:262:32
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
libc++abi.dylib: terminating with uncaught foreign exception
[1]    25745 abort      RUST_LOG=trace cargo run
```

**Expected behavior**
I expected it not to panic.

**Actual behavior**
It panicked.

**Additional context**
Add any other context about the problem here.

archseer, 2020-07-07:
Seems to be a problem with CoreBluetooth returning short UUIDs. Luckily the utils file already has a function for converting to long UUIDs.

https://github.com/deviceplug/btleplug/blob/82cd3fc7bb4ce6948a2c4f6d5e055a1cd2531d5b/src/corebluetooth/utils.rs#L50-L57

So this:

https://github.com/deviceplug/btleplug/blob/82cd3fc7bb4ce6948a2c4f6d5e055a1cd2531d5b/src/corebluetooth/central_delegate.rs#L261-L262

Should be:

```rust
let uuid = CoreBluetoothUtils::uuid_to_canonical_uuid_string(cb::attribute_uuid(s));
let uuid_str = Uuid::from_str(&uuid).unwrap();
```

I managed to get things working for my usecase by fixing a bunch of occurrences, but any call to `Uuid::from_str` under `src/corebluetooth` will need to be adjusted in the same way.

patlux, 2020-07-17:
Thanks @archseer. It works now on my Macbook Pro 16° without any problems.

---

I published the changes here: https://github.com/patlux/btleplug/commit/7fd7be9d9759b47f0736fe947b0bc656c7928546

See here for instructions how to use it: https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html

qdot, 2020-07-23:
Coming in v0.4.4 (publishing ASAP)

qdot, 2020-07-25:
0.4.4 published, closing.

PR #53 (2020-07-23): Fixes #43
Fixes #43
closed 2020-07-25 by qdot