deviceplug/btleplug #44: Update to winrt-rust

qdot, 2020-06-06:
**Feature Description**

The winrt-rs package we use for windows support is now winrt-rust, supported by Microsoft. This changes the feature handling as well as importing.

johnmave126, 2020-11-07:
I ran into a weird compilation problem recently with btleplug. After digging a little bit I found the problem arises with winrt@0.6.0

```rust
// removing this line will make it compile
use winrt;
// can also be triggered by
// use btleplug;

fn foo<T, I>(_v: &T)
where
    for<'a> &'a T: IntoIterator<Item = &'a I>,
{
}

fn main() {
    let x: Vec<bool> = Vec::new();
    // failed to compile here
    foo(&x);
}
```

Compilation error looks like:
```
error[E0275]: overflow evaluating the requirement `&_: std::iter::IntoIterator`
  --> src\main.rs:11:5
   |
3  | fn foo<T, I>(_v: &T)
   |    --- required by a bound in this
4  | where
5  |     for<'a> &'a T: IntoIterator<Item = &'a I>,
   |                    -------------------------- required by this bound in `foo`
...
11 |     foo(&x);
   |     ^^^
   |
   = help: consider adding a `#![recursion_limit="256"]` attribute to your crate (`my_test`)
   = note: required because of the requirements on the impl of `std::iter::IntoIterator` for `&winrt::comptr::ComPtr<_>`
   = note: required because of the requirements on the impl of `std::iter::IntoIterator` for `&winrt::comptr::ComPtr<winrt::comptr::ComPtr<_>>`
   = note: required because of the requirements on the impl of `std::iter::IntoIterator` for `&winrt::comptr::ComPtr<winrt::comptr::ComPtr<winrt::comptr::ComPtr<_>>>`
<more lines omitted>
```

My guess is winrt@0.6.0 did some weird trait implementation with ComPtr. Upgrading to winrt@0.7, though, doesn't trigger this compilation error.

qdot, 2021-02-23:
Updated to winrt, and now windows crates.

closed 2021-02-23 by qdot