deviceplug/btleplug #45: Remove async-std references

qdot, 2020-06-06:
**Describe the bug**
async-std was used to quickly throw the mac/windows parts of the library together. There's calls to task::spawn, which will spin up the runtime, meaning this library is not friendly for other runtimes like tokio.

Until the whole library is actually async, we should just use threads.

NZSmartie, 2021-01-18:
Looking into how to go about async in rust, if we remove `async-std` and rely on the `futures` module in the standard library and at most pull in the `futures` crate, this would make btleplug more friendly towards other runtimes.

https://rust-lang.github.io/async-book/08_ecosystem/00_chapter.html

qdot, 2021-01-18:
> Looking into how to go about async in rust, if we remove `async-std` and rely on the `futures` module in the standard library and at most pull in the `futures` crate, this would make btleplug more [...]

It's... not gonna be that easy. At all.

qwandor, 2021-01-31:
Yeah, you kind of have to commit to a runtime for things like timers to work (among other things). My preference would be for tokio over async-std, as it seems to be more widely used, but there's not really a consensus.

qdot, 2021-01-31:
> Yeah, you kind of have to commit to a runtime for things like timers to work (among other things). My preference would be for tokio over async-std, as it seems to be more widely used, but there's not really a consensus.

This isn't my experience. Buttplug runs mostly runtime agnostic, with some small utilities built in to abstract runtime specific things like task spawning, or features to swap between dependent library features when it needs to touch the network or other things. I can currently swap between threadpools, async-std/smol, tokio, and wasm-executor.

And I have timers. futures-timer is runtime agnostic too. :)

That said, I've been mostly porting to tokio's structures lately for buttplug because I just like the post v1 tokio API, but that doesn't mean this project has to pick their executor along with those structures.

qdot, 2021-01-31:
Here's the async manager model I was referring to. This is shamelessly ripped off from another library that I now can't seem to find, but I rewrote it in my library, because I didn't want to depend on the executor versions being updated in a dependency for my main library. (tokio in that branch is empty, that code is in another branch that I still need to bring in).

qdot, 2021-01-31:
https://github.com/buttplugio/buttplug-rs/tree/master/buttplug/src/util/async_manager

qdot, 2021-04-04:
This is pretty much done in dev, calling it closed for now.

closed 2021-04-04 by qdot