deviceplug/btleplug #50: Implement Debug for various types

MattesWhite, 2020-07-09:
Many management types of `btleplug` lack implementations of the `Debug` trait, e.g. `ConnectedAdapter`. This makes it inconvenience to develop a Bluetooth application as I can't simply `#[derive(Debug)]` on my state-`struct`.  The implementation is not required to include detailed information something like `ConnectedAdapter(hci0)` would be totally sufficient.

qdot, 2020-07-25:
I'm gonna be working on replacing the event system this weekend, will see what I can do about getting this done too.

qdot, 2020-07-27:
Ok, well, got ConnectedAdapter at least, so I'll close this for now and open another followup for other platforms (CoreBluetooth and UWP have some opaque type issues I need to figure out).

closed 2020-07-27 by qdot