deviceplug/btleplug #52: Unable to scan for BTLE devices with Intel 5.1 BT adapter

pretty-gay-tho, 2020-07-13:
**Describe the bug**
btleplug - via intiface-cli-rs - is unable to scan for devices using the internal, Intel BT 5.1 adapter on my laptop.

**Expected behavior**
Devices should have been discovered.

**Actual behavior**
Devices were not discovered.

**Additional context**

While trying to get intiface-desktop working with my laptop and devices, I noticed that intiface-desktop was unable to load a list of devices on my system using an internal BT adapter. After debugging in Discord, we determined that a spare BT USB dongle worked perfectly. However, I can manually scan for BTLE devices with the non-working adapter via `bluetoothctl`.

Because we were ultimately able to reproduce this with intiface-cli-rs (which uses btleplug), it was decided that the best place for this report would be here.

### System info

OS: Arch Linux
`uname -a`: `Linux janeway 5.7.7-arch1-1 #1 SMP PREEMPT Wed, 01 Jul 2020 14:53:16 +0000 x86_64 GNU/Linux`
bluez version: 5.54 (actually built from [git HEAD](https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=3ec6f74dc95cef929081be90855a104e11130150))
Non-working BT adapter info (internal BT adapter):
```
hci0:   Type: Primary  Bus: USB
        BD Address: 04:ED:33:10:1E:85  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING PSCAN ISCAN
        RX bytes:24176 acl:0 sco:0 events:907 errors:0
        TX bytes:7014 acl:0 sco:0 commands:509 errors:0
        Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'janeway'
        Class: 0x10010c
        Service Classes: Object Transfer
        Device Class: Computer, Laptop
        HCI Version: 5.1 (0xa)  Revision: 0x100
        LMP Version: 5.1 (0xa)  Subversion: 0x100
        Manufacturer: Intel Corp. (2)
```

Working BT adapter info (USB dongle):
```
hci1:   Type: Primary  Bus: USB
        BD Address: 5C:F3:70:75:6C:E8  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN ISCAN
        RX bytes:10785 acl:260 sco:0 events:368 errors:0
        TX bytes:6694 acl:260 sco:0 commands:76 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'janeway #2'
        Class: 0x10010c
        Service Classes: Object Transfer
        Device Class: Computer, Laptop
        HCI Version: 4.0 (0x6)  Revision: 0x1000
        LMP Version: 4.0 (0x6)  Subversion: 0x220e
        Manufacturer: Broadcom Corporation (15)
```

The only non-standard configuration change made to bluez (that I know of) is this, and this configuration change was not modified during any test cases. That is to say "the broken adapter was still broken and the working dongle still worked with this config change". Without this configuration change, the "broken" adapter does not see BTLE devices at all. The necessity of this change *might* be relevant, but I'm not certain:
```
~/.cache/yay/bluez-git master*
❯ grep -B3 'ControllerMode' /etc/bluetooth/main.conf
# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
ControllerMode = le
```

[... 424 lines cut ...]
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                                                               {0x0001} [hci1] 23.940874
        Address type: 0x06
          LE Public
          LE Random
        Discovery: Enabled (0x01)
> HCI Event: LE Meta Event (0x3e) plen 42                                                                                     #9 [hci1] 23.974726
      LE Extended Advertising Report (0x0d)
      
      [tons of BTLE traffic filtered out]
```

### Conclusion

|Engine|Adapter|Result|
|-|-|-|
|node|internal|fail|
|node|USB|success|
|rust|internal|fail|
|rust|USB|success|
|bluetoothctl|internal|success|

It seems that there is some difference between the BT 4.0 usb dongle and the internal, Intel 5.1 controller that causes btleplug to choke; and its interestingly reproducable on buttplug-node as well (but likely less important given that the rust engine is the glorious future of buttplug and intiface). I suspect that there is some difference in commands needed to be sent to bluez to enable bluetooth LE scanning on the Intel 5.1 controller, but I'm at the limits of my knowledge here.

However, I'm happy to help debug further and maybe to submit a patch, if we can identify what needs to be done!

MattesWhite, 2020-07-14:
I see you run the same internal Bluetooth-device like I do. Therefore, I add my information to this issue and will close mine as this probably relates to the same issue.

---
**From my original issue #51**

I'm developing a Bluetooth application for Linux that on one hand listens to broadcasted manufacturer data and on the other hand is able to connect to Bluetooth devices and execute commands.

For listening to GAP advertisements I need to constantly scan for BLE devices. When I start scanning with `ConnectedAdapter::start_scan()?` no error is raised. However, the event handler doesn't receive anything despite having a lot of Bluetooth devices in range (they are discovered using Ubuntu's utility). Using a logger I get following output:

```
Jul 09 08:41:13.952  INFO setup_bluetooth: ble_adapter::setup: enter
Jul 09 08:41:13.952  INFO setup_bluetooth: btleplug::bluez::adapter: DevInfo: HCIDevInfo { dev_id: 0, name: [104, 99, 105, 48, 0, 0, 0, 0], bdaddr: 5C:80:B6:06:7A:B0, flags: 13, type_: 1, features: [191, 254, 15, 254, 219, 255, 123, 135], pkt_type: 52472, link_policy: 5, link_mode: 32768, acl_mtu: 1021, acl_pkts: 4, sco_mtu: 96, sco_pkts: 6, stat: HCIDevStats { err_rx: 0, err_tx: 0, cmd_tx: 3656, evt_rx: 3658, acl_tx: 0, acl_rx: 0, sco_tx: 0, sco_rx: 0, byte_rx: 29650, byte_tx: 691988 } }    
Jul 09 08:41:14.078  INFO setup_bluetooth: btleplug::bluez::adapter: DevInfo: HCIDevInfo { dev_id: 0, name: [104, 99, 105, 48, 0, 0, 0, 0], bdaddr: 5C:80:B6:06:7A:B0, flags: 0, type_: 1, features: [191, 254, 15, 254, 219, 255, 123, 135], pkt_type: 52472, link_policy: 5, link_mode: 32768, acl_mtu: 1021, acl_pkts: 4, sco_mtu: 96, sco_pkts: 6, stat: HCIDevStats { err_rx: 0, err_tx: 0, cmd_tx: 3657, evt_rx: 3659, acl_tx: 0, acl_rx: 0, sco_tx: 0, sco_rx: 0, byte_rx: 29656, byte_tx: 691991 } }    
Jul 09 08:41:14.140  INFO setup_bluetooth: btleplug::bluez::adapter: DevInfo: HCIDevInfo { dev_id: 0, name: [104, 99, 105, 48, 0, 0, 0, 0], bdaddr: 5C:80:B6:06:7A:B0, flags: 13, type_: 1, features: [191, 254, 15, 254, 219, 255, 123, 135], pkt_type: 52472, link_policy: 5, link_mode: 32768, acl_mtu: 1021, acl_pkts: 4, sco_mtu: 96, sco_pkts: 6, stat: HCIDevStats { err_rx: 0, err_tx: 0, cmd_tx: 3706, evt_rx: 3707, acl_tx: 0, acl_rx: 0, sco_tx: 0, sco_rx: 0, byte_rx: 30370, byte_tx: 692752 } }    
Jul 09 08:41:14.140  INFO setup_bluetooth:Setup BLE event handler: ble_adapter::gap: enter
Jul 09 08:41:14.141  INFO setup_bluetooth:Setup BLE event handler: ble_adapter::gap: exit
Jul 09 08:41:14.141  INFO setup_bluetooth: ble_adapter::setup: exit
Jul 09 08:41:14.141  INFO DiscoveryLoop: ble_adapter::gap: enter
Jul 09 08:41:14.141 ERROR btleplug::bluez::adapter: parse error Code([55, 32, 0], MapOpt)
from: [4, 14, 4, 1, 55, 32, 0]    
Jul 09 08:41:14.141 DEBUG DiscoveryLoop: ble_adapter::gap: Start discovery
Jul 09 08:41:14.141 DEBUG DiscoveryLoop: btleplug::bluez::adapter: writing(4) [1, 11, 32, 7, 0, 16, 0, 16, 0, 0, 0]    
Jul 09 08:41:14.141 DEBUG DiscoveryLoop: btleplug::bluez::adapter: writing(4) [1, 12, 32, 2, 1, 0]    
Jul 09 08:41:14.142 DEBUG btleplug::bluez::adapter: got message HCICommandComplete(LESetScanParameters)    
Jul 09 08:41:14.143 DEBUG btleplug::bluez::adapter: got message HCICommandComplete(LESetScanEnabled { enabled: false }) 
```

I think the `ERROR` log is the root of my problem but I absolutely don't understand it.

---

My device information:

```
$> lsb_release -a
	Distributor ID:	Ubuntu
	Description:	Ubuntu 20.04 LTS
	Release:	20.04
	Codename:	focal
$> hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 5C:80:B6:06:7A:B0  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING PSCAN 
	RX bytes:19215 acl:0 sco:0 events:2864 errors:0
	TX bytes:679551 acl:0 sco:0 commands:2852 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'thinkpad'
	Class: 0x0c010c
	Service Classes: Rendering, Capturing
	Device Class: Computer, Laptop
	HCI Version: 5.1 (0xa)  Revision: 0x100
	LMP Version: 5.1 (0xa)  Subversion: 0x100
	Manufacturer: Intel Corp. (2)
```

And I also use `bluez` v5.54 build from `HEAD`.

archseer, 2020-07-28:
I'm getting a similar issue on Linux as well, the big difference seems to be that new intel chips expect `LE Set Extended Scan Enable` instead of `LE Set Scan Enable`.

archseer, 2020-07-29:
https://bugzilla.kernel.org/show_bug.cgi?id=202515

> Extended Advertising command are mutually exclusive with legacy commands and controller will reject ones if others were already issued. So if one did 'hcitool lescan' before kernel issued any EA commands, it would lead to described behavior of rejecting EA commands with 'Command Disallowed' status code.

Extended commands are part of the BLE 5 spec so we'll need to detect support somehow and then use them instead of the legacy ones.

Similar issue:

https://github.com/custom-components/sensor.mitemp_bt/issues/61

Spec for the HCI commands:

https://source.android.com/devices/bluetooth/hci_requirements#le-extended-set-scan-parameters-command

frjonsen, 2020-10-31:
Not sure if adding my comment will be of any use, but I appear to be facing the same issue as those above, with device info:

```
❯ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: C0:B6:F9:FB:F3:F0  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING PSCAN INQUIRY
	RX bytes:85924 acl:0 sco:0 events:4815 errors:0
	TX bytes:688016 acl:0 sco:0 commands:3446 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: SLAVE ACCEPT
	Name: 'redacted'
	Class: 0x2c010c
	Service Classes: Rendering, Capturing, Audio
	Device Class: Computer, Laptop
	HCI Version: 5.1 (0xa)  Revision: 0x100
	LMP Version: 5.1 (0xa)  Subversion: 0x100
	Manufacturer: Intel Corp. (2)
```

lnicola, 2021-01-21:
I have the same issue with an Intel AX 200. #105 fixes it for me.

NZSmartie, 2021-03-02:
With the merge of #105 which was released in `v0.6.0`, this issue should be now resolved.

closed 2021-03-02 by NZSmartie