deviceplug/btleplug #57: Disconnected devices won't reconnect on macOS

qdot, 2020-07-25:
**Describe the bug**

- Scan for and Connect device on macOS
- Turn device off
- Turn device back on and put in discovery mode again 

**Expected behavior**
Device is found again.

**Actual behavior**
Device is ignored.

**Additional context**
There's some logic in the CoreBluetooth module that may restrict repetition more than it should, this may be part of the problem.

blandger, 2020-07-26:
The same incorrect behavior is observed in examples/event_driven_discovery.rs

qdot, 2020-07-26:
@blandger Yeah, I think there's some expectations around the peripheral storage (something that came from rumble) that I don't quite understand, and I'm not sure what I should manage myself and what should be pushed to the user. I'm probably releasing v0.5.0 today, which overhauls the event system to using channels and starts working on this, but I think it's gonna take a few point releases to get right.

commit 5b2f1726 (2020-07-25): fix: Take peripherals off seen list on disconnect

Fixes #57
closed 2020-07-27 by qdot