deviceplug/btleplug #60: The event_driven_discovery.rs doesn work properly, may be remove that as it's not relevant any more?

blandger, 2020-07-27:
**Describe the bug**
I did small modification by adding  device lost event processing
            CentralEvent::DeviceLost(bd_addr) => {
                println!("DeviceLost: {:?}", bd_addr);

and that looks like program prints 'Discovered event' only after turning on BLE device.
Is 'DeviceLost' event not fired? I see description in comments:
> The first time the getter/ is called, it will return Some(Receiver<CentralEvent>). After that, it  will only return None

If that example does not works properly so probably it's worth deleting?

**Expected behavior**
I expected to see correct Discovered/Lost events processed for turned on/off device

**Actual behavior**
Program prints 'Discovered event' only after turning on BLE device.
When I try turn on/off device several times on Linux, it doesn't work, only one event if fired.

**Additional context**
Linux Mint 20, x64

qdot, 2020-07-27:
There's no reason to delete the example. It still shows how the event system works.

How do you expect a device to be discovered if it isn't on? I don't understand what you're asking there.

As for DeviceLost, that event is currently never fired, and may need to be removed (I'm not sure about platform capabilities for realizing that a device is no longer being advertised). What's happening is that a device is seen, stored in the peripherals array, then when a new advertisement is seen, the DeviceUpdated event is fired. The example should be readjusted to make that clearer.

DeviceDisconnected will never be fired because there is no connect call being made.

blandger, 2020-07-27:
Agree about DeviceConnected/DeviceDisconnected will never fire in that example.

About DeviceDiscovered / DeviceLost ... It seems to me, there was some kind of 'device disconnected' event in Android library, if I'm not mistaken. So that is why I expected to see DeviceLost to be pair of DeviceDiscovered event.

Couple questions...

1. 
**Why UUID::B128 has reversed internal representation?**
impl Display for UUID {
......for i in (1..a.len()).**rev()** {

impl FromStr for UUID {
   **bytes.reverse();**

const RECEIVE_CHANNEL_CHARACTERISTIC: UUID = UUID::B128([0x1B, 0xC5, 0xD5, 0xA5, 0x02......]);

I was trying to create example with subscription to Characteristic with some known UUID (from one of my previous Android experience/example) and was confused trying to match it with 'known UUID', like :

```
...for char_item in chars_vector.iter() {
... if char_item.uuid == RECEIVE_CHANNEL_CHARACTERISTIC 
                && char_item.properties == CharPropFlags::NOTIFY 
```
and was confused about 'reversed' order after debugging. That was not obvious...

In general all I wanted just get Char constant from string like:
    UUID::from_str("6E400001-B534-F393-68A9-E50E24DCCA9E");
but as I see ::from_str is not public now and doesn't support such formatting.
(I can try to add additional method like ::parse_from_UUID(...) for such string variant.)

2.
I still didn't succeeded on creating example with Subscribe / Unsubscribe from Characteristic (Sync) for getting continuous/stream data from EEG device. I'll keep trying but still luck of minimal example. lights.rs is not quite good.
Also I will try to read more about tokio stuff. May be that will help in that too.

qdot, 2020-07-27:
For DeviceLost missing, it's one of the problems with having a VERY cross-platform library. I'm hindered by working with what ALL platforms support, and right now we don't have Android support at all, and I'm not sure if desktop platforms can tell what losing a device means. The plan right now is to do a sweep of all platform specific APIs when building the new async layer (which will be btleplug v1), I'll hopefully be able to gather more info then.

> Agree about DeviceConnected/DeviceDisconnected will never fire in that example.

We could definitely use better examples, it's mostly just a problem of figuring out what devices would be best to test since BLE is so widespread.

> **Why UUID::B128 has reversed internal representation?**

This came from rumble, the original device this library forked off of. We need a more extensible UUID type, I just haven't had time to add it.

> 
> I still didn't succeed on creating example with Subscribe / Unsubscribe from Characteristic (Sync) for getting continuous/stream data from EEG device. I'll keep trying but still luck of minimal example. lights.rs is not quite good.

Subscriptions and notifications are a mess due to the way this library handles callbacks. These should ideally be channels/streams, similar to the events. They do at least work right now. If you call .subscribe() on a characteristic, and have set a notification handler, you do get info back. I have this working on https://github.com/buttplugio/buttplug-rs, but it's in a fairly complicated bit of code that's not real clear to cut out as an example.

> Also I will try to read more about tokio stuff. May be that will help in that too. 

I wouldn't worry too much about Tokio specifically here, more just getting familiar with futures in general. As this is a library, we really need an runtime agnostic implementation, which can get very tricky. I have structures to handle this in buttplug-rs already (we handle tokio/async-std/smol/thread pools), I plan on porting them over here once I start looking at the async redesign.

blandger, 2020-07-28:
> ...I'm not sure if desktop platforms can tell what losing a device means. 
I thought the loosing connection to BLE (stopped data streaming) could be one of obvious choices, in case device was connected and that is possible by technical reasons. But probably that is not easy.

> The plan right now is to do a sweep of all platform specific APIs when building the new async layer (which will be btleplug v1), 
I'm aware not many unit/integration tests in code, but as I understand that is not an easy task to do properly without real devices. I would say that is not simple task even having devices locally. Mocking some stuff is a choice for limited number of specific cases only.

> We could definitely use better examples, it's mostly just a problem of figuring out what devices would be best to test since BLE is so widespread.
Do you have such experience in real (kind of production)? I've heard about 'device farm' practice in big companies when they have a bunch of Android test devices ('farm') connected to some 'host server' and continuously running tested applications/test on them. Unfortunately that way is expensive and probably can't be applied on project like this one.

> > **Why UUID::B128 has reversed internal representation?**
> This came from rumble, the original device this library forked off of. We need a more extensible UUID type, I just haven't had time to add it.
I can work on that, because I did one attempt locally, but not finished it by tech reason. I'll give it one more try, because 'reversed order' for UUID confused me a lot previously.

> ...If you call .subscribe() on a characteristic, and have set a notification handler, you do get info back. I have this working on https://github.com/buttplugio/buttplug-rs, but it's in a fairly complicated bit of code that's not real clear to cut out as an example.
I tried that but didn't succeed with my device. Thanks, I will look at examples in:  /buttplug-rs/buttplug/examples/

> I wouldn't worry too much about Tokio specifically here, more just getting familiar with futures in general. 
Sure, agree. It's only for reference as that is 'most full' and 'quite complex' reference implementation. Also learning tokio is helpful in usage by future server-side projects.

> ....I plan on porting them over here once I start looking at the async redesign.
Do you have any time lines for such plans and possible time to finish draft version at least? I'm just curious.

qdot, 2020-07-28:
> 
> 
> > ...I'm not sure if desktop platforms can tell what losing a device means.
> 
> I thought the loosing connection to BLE (stopped data streaming) could be one of obvious choices, in case device was connected and that is possible by technical reasons. But probably that is not easy.

Yeah, I'm just not sure if all desktop APIs differentiate between disconnection due to stream loss versus request. Will have to see.

> I'm aware of (not many) unit/integration tests in code, but as I understand that is not an easy task to do properly without real devices. I would say that is not simple task even having devices locally. Mocking some stuff is a choice for limited number of specific cases only.

Yup. Could always use more tests, but I'm also not super motivated to work on the codebase as is. Would like to implement those as part of the new system I have planned.

> Do you have such experience in real (kind of production)? I've heard about 'device farm' practice in big companies when they have a bunch of Android test devices ('farm') connected to some 'host [...]

This project will get plenty of exercise from the users of my other libraries. I do have experience with device farms (I was one of the first engineers on FirefoxOS), but this is a single developer, open source project. A device farm is not happening here.

> I tried that but didn't succeed with my device. Thanks, I will look at examples in: /buttplug-rs/buttplug/examples/

Examples aren't going to help you much there, as the only subscription system happens far back in our protocol implementations at the moment. server/comm_managers/btleplug/btleplug_internal.rs is where subscriptions and notification handling happen, but it's a badly built state machine set very deep in the guts of the library, so not sure how useful it'll be.

> Do you have any time lines for such plans and possible time to finish draft version at least? I'm just curious.

Ehhhhhh hopefully in the next 4-6 weeks to start? Finishing, not sure. I have a dayjob *and* run a startup, so I'm kinda stretched thin as is.

MattesWhite, 2020-07-29:
> Why UUID::B128 has reversed internal representation?

Just for completeness I want to add that `BDAddr` is also internally stored in reverse order.

blandger, 2020-07-29:
> > Why UUID::B128 has reversed internal representation?
> Just for completeness I want to add that `BDAddr` is also internally stored in reverse order.

Thank you for pointing on that. I will consider that too.

> I don't have much time either but I would like to help you. Maybe you can open some _help wanted_ issues (like _improve UUID and/or BDAddr implementation_).
Do you think it is good idea to work on the same issue? Or may be I didn't catch you... I suppose that's not a problem to create issue and ask to assign it to me (or you), as I think.

MattesWhite, 2020-07-29:
No there should always only be one to work on one issue. However, there shouldn't be a problem working together on the same project. As long as there is one that coordinates the issues and resolves conflicts (even there shouldn't be one). For example, it shouldn't be a problem to rework the implementation of UUIDs while @qdot is working on `async/.await`.

I'm just a little bit confused on the project structure seeing that @blandger is the author. Who is actually in charge of this project given that @qdot seems to do the most work.

blandger, 2020-07-29:
Sure, qdot is Kyle Machulis. He is repo owner and main contributor. I did only couple PRs with example.
I wanted work on UUID, because I stumble on it trying to create 'subscribe on BLE data example', that's why I asked about it.
Pls find something else in issues or... think about another stuff to work on.

qdot, 2020-07-29:
@MattesWhite Thanks for the offer of help! Right now the main problem is figuring out how much work to ask for help on versus waiting to start the API overhaul then asking once I actually have a design doc in place. I'll definitely be marking stuff Help Wanted once I figure out what can be done that won't be stomped on by new API changes quickly.

@blandger If you want to work on something, please file an issue, describe what you want to fix, and ask for assignment. Please don't fight other people interested in the project who just want to help. While I did fork this code from multiple places, I run this project and have my own plans which I want to make sure are followed.

blandger, 2020-07-29:
Kyle, no problem. probably I was quite polite.
/deviceplug/btleplug/issues/68

NZSmartie, 2021-01-11:
I've made sure that the PR #105 correctly fires the DeviceLost event when BlueZ reports that the device has been removed

qdot, 2021-07-31:
Ok, somehow we've got from DeviceLost firing for no one to DeviceLost only firing on Bluez to DeviceLost only firing on macOS. I'm gonna close this out and create a new bug (https://github.com/deviceplug/btleplug/issues/176) to implement DeviceLost on Linux (again :| ) and Windows (if that's even possible?).

PR #105 (2021-02-04): BlueZ: Replace raw HCI sockets with BlueZ over DBus
This is a complete rewrite of the "bluez" platform specific code.

A lot has happened and I did my best to commend code to explain how/why a well as keep commits subjective so that changes can be understood in context.

Things I want to do before merging:
- [ ] ~~Verify I didn't break the existing API~~ (Maintaining a backwards compatible API is a lot harder than I thought and ends up with a lot of messy dead code)
- [ ] Ensure both passive and active discovery work as intended
- [x] Ensure the code isn't too terrible looking
- [ ] Test this  [... cut]
closed 2021-07-31 by qdot