deviceplug/btleplug #63: CoreBluetooth: peripheral.command actually sends a write request

archseer, 2020-07-28:
`peripheral.command` seems to send a `write request` which slows throughput because it waits for an ACK.
https://github.com/deviceplug/btleplug/blob/69d341b541bce944b6577b9cd85966c2c7d416b4/src/corebluetooth/peripheral.rs#L215

`peripheral.request` itself seems to be a placeholder.

The write value wrapper seems to specify `type: 0` which indicates `withResponse`, resulting in a request. https://github.com/deviceplug/btleplug/blob/5003774aaa810dc39c81943107e425ed75be00d4/src/corebluetooth/framework.rs#L370

https://developer.apple.com/documentation/corebluetooth/cbcharacteristicwritetype

The same function could be used for `request` with type set to `withoutResponse`.

qdot, 2020-07-28:
Oh, huh, thanks! Totally didn't catch that, and it's in some of the Servo code I just kinda brought in without checking. Will see about getting this fixed in a point release soon.

archseer, 2020-07-30:
Since the code already assumes this is a write without response, it immediately returns an OK anyway:

https://github.com/deviceplug/btleplug/blob/69d341b541bce944b6577b9cd85966c2c7d416b4/src/corebluetooth/internal.rs#L341-L350

I ended up just flipping the type to 1:
https://github.com/archseer/btleplug/commit/cf73f7327cee4d71e746be6cb9a06d0264af76d2

Sidenote: the OSX behavior is a bit variable: If you try sending a value longer than the negotiated MTU it'll silently flip to using write requests again.

I figured out the negotiated MTU size for my specific hardware using a packet sniffer, but I'd like to expose that as a property on the peripheral (being able to negotiate the size would be good too but not as crucial since most platforms automatically negotiate the largest supported size by the hardware)

qdot, 2021-07-31:
Closing out as this is now invalid as of v0.8, we've change the UI to just use write and a Write mode.

closed 2021-07-31 by qdot