deviceplug/btleplug #66: Consider using rustfmt

archseer, 2020-07-29:
I've got my editor set to auto format, so changing any file in this project completely reformats the file.

If you don't like using rustfmt, another option would be adding a `.rustfmt.toml` that disables the formatter: https://github.com/BurntSushi/ripgrep/commit/688003e51c06f1eab7ebd72a9d0435f81b44002f (though even ripgrep now [switched over](https://github.com/BurntSushi/ripgrep/blob/master/rustfmt.toml)).

qdot, 2020-07-29:
tl;dr: Yeah no prob.

Not tl;dr: I'm a big fan of rustfmt, use it on my other projects. So far I hadn't touched this one because I'm trying to limit my changes to this version of the codebase before starting on v1 (which I should really explain in the README), but at this point if I'm gonna have people contributing, it'd be nice for it to be cleaner, so I'll get that run before the 0.5.1 release.

qdot, 2020-08-04:
Ran fmt as part of 0.5.1. Now to address all those clippy errors so maybe we can have a fully green build for once.

closed 2020-08-04 by qdot