deviceplug/btleplug #68: Rework BDAddr + UUID structs representaion

blandger, 2020-07-29:
**BDAddr + UUID structs representaion**

I would like to rework BDAddr + UUID structs internal 'reversed' representation by removing that piece of code (in format/display/debug + parse parts). It confuses a lot. Also I would like to add parse UUID as string variant(s).

BDAddr might be stored as MAC:
https://www.novelbits.io/bluetooth-address-privacy-ble/

characteristic for UUID might be:
_A universally unique identifier (UUID) is a 128-bit (16 bytes) number_
https://learning.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html

MattesWhite, 2020-07-30:
To clarify I don't wanted to be the one refactoring `UUID`s and `BDAddr`, I just offered my help if it is appreciated. It is just the case that I have similar problems in my use case. I didn't felt offended, everything is fine.

Now back to the issue.

---

As you already noted about format/display/debug and parse, I'd like to give my list of requirements/features/wishes:

- Implement not only `Display` but also other [`std::fmt` traits](https://doc.rust-lang.org/std/fmt/#traits) like `LowerHex` and `UpperHex`.
- The default display should be with delimiters, i.e. `BDAddr` with `:` and UUID with hyphens.
- Optional formatting functions for serialization without delimiters.
  - This is something I used quite often as I use `BDAddr` and `UUID`s in the routes of my HTTP-server.
- The same for parsing: The parsers should not rely on the presence of delimiters and be case insensitive for hex digits.

This is only a list of what I would have tried to do. I don't insist on them and I'm open to discussion. Of course I will respect the wishes/vision of the project owner :wink: .

blandger, 2020-07-30:
Good additions, Mattes. I don't mind you can take that issue. Let Kyle decide by himself.

I can take issue upgrading project to NOM 5.1 version. I tried that locally, but it needs more research, because difference between 4 and 5 quite significant.

qwandor, 2021-03-10:
@MattesWhite's refactor of BDAddr is now merged, is an improvement.

However, we could consider instead using one of the existing crates which provides a standard MAC address type, such as [eui48](https://crates.io/crates/eui48) or [macaddr](https://crates.io/crates/macaddr). I haven't had a chance yet to see which of the two is a better choice. Any opinions?

MattesWhite, 2021-03-11:
Well, BDAddrs are not that much code. The crates you mention mainly target the MAC addresses from IP protocol and their conventions, e.g. check if a MAC address is for broadcasting. I think the overlap of both use cases for MAC addresses is rather small.

The only benefit I see from using `eui48` (`macaddr` is rather small) is having more sophisticated serializing and parsing, e.g. recognize hyphens as delimiters. However, both don't support _my_ `_no_delim` serialization/parsing. There is [`to_hexadecimal()`](https://docs.rs/eui48/1.1.0/eui48/struct.MacAddress.html#method.to_hexadecimal) in `eui48` but that adds some nasty `0x` upfront. In the end, I think introducing a MAC crate won't add much benefit over the current implementation.

The only interesting feature still left is the `serde` serialization of a BDAddr. Currently we use the default serialization due to the definition of BDAddr this currently leads to, e.g. JSON, serialization like this: `{ "address": [0, 1, 2, 3, 4, 5] }` which is not very handy. `macaddr` is roughly the same. `eui48` usually serializes to `00-01-02-03-04-05` and with a feature this can be changed to use colons as delimiters. I think this isn't a good solution, either. My personal preference would be to default to colon delimited BDAddr and optional de-/serializer functions [like in `chrono`](https://docs.rs/chrono/0.4.19/chrono/serde/index.html) to allow to de-/serialize from `_no_delim` format or from an array of bytes (or other formats in the future).

qdot, 2021-07-31:
This got done in v0.8, closing.

closed 2021-07-31 by qdot