deviceplug/btleplug #69: Consider supporting blurz or other dbus interface to bluetoothd

rvolgers, 2020-08-01:
btleblug currently takes the approach of directly using a hci socket to communicate with the kernel's bluetooth stack on Linux. An alternative approach is to use dbus to communicate with bluetoothd. It's probably worth at least discussing the trade-offs, and potentially adding support for the alternative approach.

Advantages of dbus:
* Much, much, *much* simpler, pretty much all logic is implemented in existing libraries instead of getting deep into HCI packet parsing in btleplug itself.
* (TODO: verify this) it seems like the dbus API might be more reliable, possibly because it caches information

Disadvantages:
* Dependency on separate daemon and dbus. I'm not sure where it is installed or not. It was present on my raspberry pi.
* If you use libdbus, cross compiling becomes harder (you need a custom docker container for cross). You could of course use a pure Rust dbus library (zbus), but that may cause longer compile times.

rvolgers, 2020-08-01:
Um, I don't remember *assigning* this to anyone, but I assume it's just how new issues work...

qdot, 2020-08-01:
First off, yup, everything here auto-assigns to me, just means I have less of a chance of losing track of it.

As for DBus, heh, I just closed #9 because I figured async'ing the sockets would just be a matter of using a generic async fd type. That said, since the system is extensible, it'd definitely be worth exploring this if someone feels up to it. I am admittedly weakest on bluez (both socket and dbus), and my very vague recollection was that the dbus API was missing... something? But this is from years ago, so if someone that's up on the current state knows more, would love to have more info.

oxc, 2020-10-13:
The DBus API can not, for example, explicitly initiate a passive scan. You can use the mgmt-api to add devices to the drivers whitelist, for which the kernel will scan passively. In both APIs, there is no such thing as "starting a passive scan".

The mgmt-api is not available via dbus, but requires opening a HCI (control) socket.

It might still be worth combining those two APIs, since the HCI control socket should be much easier to handle than the raw sockets.

NZSmartie, 2020-12-31:
I'm currently making progress on this, using `dbus-rs`. I've got adapter and peripheral discovery working 
https://github.com/NZSmartie/btleplug/tree/feature/bluez-dbus

NZSmartie, 2020-12-31:
as for the management socket, that still requires a setcap admin capability. Looking at the bluez source code for 5.54, passive scanning does seem to happen, but it is not explicit. I'll poke around with it a bit more once i've got my fork of the bluez APIs sorted.

qwandor, 2021-01-01:
I'm interested in potentially helping with this. I've ended up making my own library wrapping the BlueZ D-Bus API (currently at https://github.com/alsuren/mijia-homie/tree/master/mijia/src/bluetooth as a module in another crate, but I would like to split it out into a standalone crate) with a decent async API, and from what I can see from poking around the btleplug source so far, it shouldn't be too hard to implement most of btleplug on Linux based on that, with some wrappers to make everything blocking. It would be nice to see btleplug expose a proper async API though.

NZSmartie, 2021-01-06:
@qwandor I'll keep an eye on your project, as it may be more mature and better built than my attempt to port this library across to BlueZ over DBus. I'm not yet familiar with async in Rust, and I'm mostly retrofitting the existing API to work with DBus so that there are no breaking API changes for end users. 

I'll be happy to scrap my work and integrate yours as a dependency if it works out well! 

As for the API design, this library does appear to need and want an overhaul. So that could be the next attempt to help improve this project, input from everyone will be greatly appreciated

qwandor, 2021-01-11:
@NZSmartie I've released https://crates.io/crates/bluez-async, in case that's any use to you.

qdot, 2021-02-05:
This is done in v0.6, closing! Thanks @NZSmartie :D

closed 2021-02-05 by qdot