deviceplug/btleplug #72: Error running example discover_adapters_peripherals: 'main' panicked at 'How did we get here without a handle?'

cassc, 2020-08-13:
**Describe the bug**
When running the example `discover_adapters_peripherals`, the program panicked with  panicked with the error message: `'How did we get here without a handle?`

**Actual behavior**

```
sudo ./target/debug/examples/discover_adapters_peripherals
connecting to BLE adapter: ...
connected adapter "hci0" is UP: true
adapter states : {Up, Running}
peripheral : None is connected: false
SKIP connect to UNKNOWN peripheral : 62:E0:4B:59:37:D0 properties: PeripheralProperties { address: 62:E0:4B:59:37:D0, address_type: Random, local_name: None, tx_power_level: Some(12), manufacturer_data: Some([76, 0, 16, 5, 80, 28, 47, 95, 62]), discovery_count: 14, has_scan_response: true }, characteristics: {} 
peripheral : None is connected: false
SKIP connect to UNKNOWN peripheral : 55:E0:3C:1F:A2:D1 properties: PeripheralProperties { address: 65:A0:3C:EF:B2:D1, address_type: Random, local_name: None, tx_power_level: Some(12), manufacturer_data: Some([76, 0, 16, 5, 3, 24, 253, 115, 195]), discovery_count: 1, has_scan_response: false }, characteristics: {} 
peripheral : Some("c_Controller") is connected: false
start connect to peripheral : Some("bell_Controller")...
thread 'main' panicked at 'How did we get here without a handle?', src/bluez/adapter/acl_stream.rs:254:22
```

**Additional context**
Program running on Arch Linux with Rust 1.45.2

lnicola, 2020-09-03:
Maybe it's not your case, but is it possible that this might happen if an application subscribes to notifications from a device, gets stopped (e.g. Ctrl-C), then starts again?

qdot, 2021-07-31:
We ripped out the BlueZ core in deference to dbus in v0.8, so this should be all fixed.

closed 2021-07-31 by qdot