deviceplug/btleplug #75: Missing local_name on device discovery event

Beetix, 2020-09-04:
**Describe the bug**
local_name is None on device discovery event

**Expected behavior**
local_name to be defined

**Actual behavior**
local_name is defined on device update event only

**Additional context**
Using event driven discovery example with the following event handling code:
```rust
while let Ok(event) = event_receiver.recv() {
    let addr;
    match event {
        CentralEvent::DeviceDiscovered(bd_addr) => {
            println!("Discovered {}", bd_addr);
            addr = bd_addr;
        },
        CentralEvent::DeviceLost(bd_addr) => {
            println!("Lost {}", bd_addr);
            addr = bd_addr;
        },
        CentralEvent::DeviceUpdated(bd_addr) => {
            println!("Updated {}", bd_addr);
            addr = bd_addr;
        }
        CentralEvent::DeviceConnected(bd_addr) => {
            println!("Connected {}", bd_addr);
            addr = bd_addr;
        }
        CentralEvent::DeviceDisconnected(bd_addr) => {
            println!("Disconnected {}", bd_addr);
            addr = bd_addr;
        }
    }
    let peripheral = central.peripheral(addr).unwrap();
    let name = peripheral
        .properties().local_name;
    match name {
        Some(name) => println!("Name {}", name),
        None => println!("No name !"),
    }
};
```
Output:
```
Discovered A4:C1:38:53:88:21
No name !
Updated A4:C1:38:53:88:21
Name LYWSD03MMC
```

qdot, 2020-09-05:
What platform is this on, and are you sure the name is coming in the first advertising packet seen?

Beetix, 2020-09-05:
This is on Arch Linux x86_64.

When I use `btmon` I get the following:
```
> HCI Event: LE Meta Event (0x3e) plen 33                                                       #9 [hci0] 11.707437
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: A4:C1:38:53:88:21 (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 21
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Service Data (UUID 0xfe95): 30585b050121885338c1a4280100
        RSSI: -60 dBm (0xc4)
> HCI Event: LE Meta Event (0x3e) plen 24                                                      #10 [hci0] 11.709410
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: A4:C1:38:53:88:21 (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 12
        Name (complete): LYWSD03MMC
        RSSI: -60 dBm (0xc4)
@ MGMT Event: Device Found (0x0012) plen 47                                               {0x0002} [hci0] 11.709428
        LE Address: A4:C1:38:53:88:21 (Telink Semiconductor (Taipei) Co. Ltd.)
        RSSI: -60 dBm (0xc4)
        Flags: 0x00000000
        Data length: 33
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Service Data (UUID 0xfe95): 30585b050121885338c1a4280100
        Name (complete): LYWSD03MMC
```
So the name comes from the scan response. Is the device discovery event generated on the first packet received?

qdot, 2021-07-31:
Device discovered is sent on the first packet received, which may or may not be an advertisement. It's best to watch for both DeviceDiscovered and DeviceUpdated, and if you don't get a name in DeviceDiscovered, just expect one in an Updated event later.

closed 2021-07-31 by qdot