deviceplug/btleplug #82: Device.characteristics() is empty

aklajnert, 2020-09-28:
This is an example from the `README`:  
```rust
light.discover_characteristics().unwrap();

// find the characteristic we want
let chars = light.characteristics();
let cmd_char = chars.iter().find(|c| c.uuid == UUID::B16(0xFFE9)).unwrap();
```

In my case, `light.characteristics()` is empty (at least on Windows).  

It took me some time to figure out that characteristics can be obtained directly from the `dicsover_characteristics()` method:  
```rust
let cmd_char = light.discover_characteristics().unwrap().iter().find(|c| c.uuid == UUID::B16(0xFFE9)).unwrap();;
```

In my case, I'm connecting to the IKEA desk instead of Bluetooth light, but it shouldn't make any difference here.  

I didn't test that on Linux.

chrisharper77, 2020-11-12:
Hello,

Unfortunately I'm having the same issue in windows. I have been unable to confirm in linux (but I'm working on it).

I am a little confused about your second statement up there. the first unwrap creates a borrowed value error when I gave that a shot . I'm a bit confused  is the implication that that statement works to get the characteristics in windows for you?

aklajnert, 2020-11-12:
Yeah, the `discover_characteristics()` method returns the characteristics data for me, but they are not accessible by calling `characteristics()`.

johnmave126, 2020-11-12:
I guess this is related:
https://github.com/deviceplug/btleplug/blob/master/src/winrtble/peripheral.rs#L235-L241

ShayBox, 2021-03-28:
I have run into this myself, works on Linux but not Windows
https://github.com/ShayBox/Lighthouse/blob/master/src/main.rs#L101-L105

qwandor, 2021-04-01:
It looks like the `Peripheral.characteristics` field is never set on Windows, but `Peripheral.ble_characteristics` is. It should be possible to use that instead. I'll have a look at it this weekend.

qwandor, 2021-04-01:
@ShayBox @aklajnert @chrisharper77, can you let me know if #142 fixes the problem for you?

aklajnert, 2021-04-02:
@qwandor - I'm not sure if I'm doing something wrong, but this branch doesn't work for me. I took the example code from the docs, with slight modifications:

```rust
use std::thread;
use std::time::Duration;
#[cfg(target_os = "linux")]
use btleplug::bluez::{adapter::Adapter, manager::Manager};
#[cfg(target_os = "windows")]
use btleplug::winrtble::{adapter::Adapter, manager::Manager};
#[cfg(target_os = "macos")]
use btleplug::corebluetooth::{adapter::Adapter, manager::Manager};
use btleplug::api::{Central, Peripheral, WriteType};

const DEVICE_NAME: &str = "Desk";

pub fn main() {
    let manager = Manager::new().unwrap();

    let adapters = manager.adapters().unwrap();
    let central = adapters.into_iter().nth(0).unwrap();

    central.start_scan().unwrap();
    thread::sleep(Duration::from_secs(2));

    let device = central.peripherals().into_iter()
        .find(|p| p.properties().local_name.iter()
            .any(|name| name.contains(DEVICE_NAME))).unwrap();

    device.connect().unwrap();

    println!(
        "discover_characteristics(): {:?}",
        device.discover_characteristics()
    );
    println!("characteristics(): {:?}", device.characteristics());
}

```

If I'll use code from the master, everything works as described here. Dependencies:  
```toml
[dependencies]
btleplug = { git="https://github.com/deviceplug/btleplug.git" }
```

Output:  
```
discover_characteristics(): Ok([Characteristic { start_handle: 0, ...
characteristics(): {}
```

Now, when I change dependencies to the one below, the code doesn't compile:
```toml
[dependencies]
btleplug = { git="https://github.com/deviceplug/btleplug.git", branch="characteristics" }
```

```rust
error[E0603]: module `winrtble` is private
   --> src\main.rs:12:15
    |
12  | use btleplug::winrtble::manager::Manager;
    |               ^^^^^^^^ private module
    |
note: the module `winrtble` is defined here
   --> C:\Users\aklajnert\.cargo\git\checkouts\btleplug-4193de4d1cdc37dd\1811316\src/lib.rs:105:1
    |
105 | mod winrtble;
    | ^^^^^^^^^^^^^
```

I don't understand why the module is now private, in the PR I can't see any change that would do that...

qwandor, 2021-04-02:
Ah, a bunch of things are changed in the dev branch compared to master. Have a look at the examples from the dev branch, e.g. https://github.com/deviceplug/btleplug/blob/dev/examples/lights.rs.

For the specific error you're seeing, you can now just do `use btleplug::platform::{Adapter, Manager, Peripheral};` and you'll get the appropriate type for the current platform, rather than having to do all the `#[cfg(target_os = "windows")]` stuff.

aklajnert, 2021-04-02:
Oh, I didn't notice that the branch is against `dev`. Now both `discover_characteristics()` and `characteristics()` seem to work correctly.

qwandor, 2021-04-02:
Great, thanks for testing! I'll merge it into dev.

closed 2021-04-02 by qwandor

AsterisMono, 2024-08-13:
Hi! I'm having the same issue on Windows (Windows 10 19045, rustc 1.79.0, btleplug = "0.11.5").

One available workaround is to run `device.discover_services().await?;` before trying to access `device.characteristics()`, but other platforms (I've tested MacOS by far) do not require this.

PR #142 (2021-04-02): Fix Peripheral::characteristics() on Windows
This should fix #82.