deviceplug/btleplug #83: Create struct with `Peripheral` on Linux

djent-y, 2020-10-04:
This is more a question than an actual issue. I need a `Peripheral` struct within my own struct. On Windows it is easy, as the underlying struct is public. On Linux I need to use the public trait. It seems to work [internally](https://github.com/deviceplug/btleplug/blob/master/src/api/adapter_manager.rs#L20), but I can't make the code compile.  

What am I doing wrong here?

```rust
use btleplug::api::{BDAddr, Central, Peripheral};
use btleplug::bluez::manager::Manager;
use btleplug::Error;
use std::str::FromStr;

// cannot import the Peripheral struct as the module is private
// use btleplug::bluez::adapter::peripheral::Peripheral as PT;

struct MyStruct<PeripheralType: Peripheral> {
    device: PeripheralType,
}

impl<PeripheralType> MyStruct<PeripheralType>
where
    PeripheralType: Peripheral,
{
    fn get_device() -> PeripheralType {
        let central = Manager::new()
            .unwrap()
            .adapters()
            .unwrap()
            .into_iter()
            .next()
            .unwrap()
            .connect()
            .unwrap();
        central
            .peripheral(BDAddr::from_str("2A:00:AA:BB:CC:DD").unwrap())
            .unwrap()
    }

    pub fn new() -> Self {
        let device = Self::get_device();
        Self { device }
    }
}

fn main() -> Result<(), Error> {
    let _ = MyStruct::new();

    Ok(())
}
```

The error I'm getting:  
```rust
error[E0308]: mismatched types
  --> src/main.rs:27:9
   |
13 |   impl<PeripheralType> MyStruct<PeripheralType>
   |        -------------- this type parameter
...
17 |       fn get_device() -> PeripheralType {
   |                          -------------- expected `PeripheralType` because of return type
...
27 | /         central
28 | |             .peripheral(BDAddr::from_str("2A:00:AA:BB:CC:DD").unwrap())
29 | |             .unwrap()
   | |_____________________^ expected type parameter `PeripheralType`, found struct `btleplug::bluez::adapter::peripheral::Peripheral`
   |
   = note: expected type parameter `PeripheralType`
                      found struct `btleplug::bluez::adapter::peripheral::Peripheral`
```

I'm not experienced in Rust so I don't understand why it doesn't work in my case.

ChrisCA, 2020-10-04:
@djent-y 
We had a similar discussion in #73 
You might want to have a look at this and find it useful. :-)

djent-y, 2020-10-04:
@ChrisCA - I did see that discussion but it seems unresolved. I still don't know how to adapt my example to compile.

qdot, 2021-07-31:
Everything is pub'd in v0.8, so I think we should be good here.

closed 2021-07-31 by qdot