deviceplug/btleplug #87: Implement `read_by_type()` on Windows

aklajnert, 2020-10-07:
**Feature Description**

Currently the `read_by_type()` method in `src/winrtble/peripheral.rs` has exactly the same implementation as simple `read()`. The additional `_uuid` argument is unused, while the Linux implementation seems to support it.

qwandor, 2021-02-04:
Hi there, I'm looking into `read_by_type` and trying to figure out whether to remove it entirely or change it somehow. Can you explain what you are trying to use it for?

aklajnert, 2021-02-05:
Hi, I'm far from being an expert on the Bluetooth protocol, but I'm trying to control the IKEA Idasen desk from Rust. While in general, I can do that successfully, the desk sometimes is "stuck" by constantly reporting back the same height even if it is changed manually. The problem is gone if I manipulate the desk by using any other implementation found on GitHub (none of them is in Rust) - then the desk starts reporting the correct position again.  
The only difference between my implementation and others is that they are sending two `UUIDs` on each message instead of just one.

qdot, 2021-02-05:
@aklajnert Can you post one of the non-rust versions that you've gotten working? We can probably translate the requirements from that.

aklajnert, 2021-02-05:
@qdot - now I took another look at the other working projects (https://github.com/newAM/idasen/blob/master/idasen/__init__.py) and they seem to use one `UUID` for reading. It seems like I've messed up something else.  

Do you have any suggestions on how to debug such an issue? Do you have any good method for sniffing the Bluetooth communication on Windows? I remember I've tried to do it before, but failed.

qwandor, 2021-02-06:
Yeah, that code seems to be just doing normal writes to and reads from GATT characteristics, you shouldn't need `read_by_type` for any of that. I don't have any good ideas for debugging. Does it work on any other platforms?

aklajnert, 2021-02-09:
>  Does it work on any other platforms?

It compiles, but I don't have any other platform than Windows to test it. Unfortunately, Bluetooth doesn't work on WSL.

aklajnert, 2021-02-24:
I think I've figured it out - it seems that I have to call `subscribe()` first before reading a characteristic. When I do that it seems to work as expected, and the values are updated correctly when I move the desk.  
I'm not 100% sure if that's the case, as the problem is not really deterministic and it seems to fix itself for a while to come back later, anyway that's probably not due to lack of `read_by_type()` functionality.

closed 2021-02-24 by aklajnert