deviceplug/btleplug #89: Implement MTU change

MattesWhite, 2020-10-16:
**Feature Description**

The Linux version of `btleplug` does currently not support the change of the MTU. 

see: https://github.com/deviceplug/btleplug/blob/master/src/bluez/adapter/acl_stream.rs#L225

qdot, 2021-02-05:
@NZSmartie What's the status of MTU changes in the bluez stuff?

NZSmartie, 2021-02-05:
BlueZ handles the negotiation of MTU automatically and isn't directly exposed through the DBus API. 
There is mention of MTU in https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt, but that applies to the `AcquireWrite` and `AcquireNotify` which isn't used in btleplug, yet. 
Since the code linked above relates to data being received with an MTU that doesn't match that of btleplug, it is no longer relevant as that is now all handled by BlueZ.

qdot, 2021-02-05:
Ok, closing this out then.

closed 2021-02-05 by qdot