deviceplug/btleplug #91: Unwrapped result does not propagate the error

Gui-Yom, 2020-11-06:
**Describe the bug**
The library calls unwrap on an error result.
```rust
if adapter.start_scan().is_err() {  // <-- HERE on start_scan()
    println!("Please enable bluetooth");
    exit(-1);
}
```
Bescause of this in [watcher.rs](https://github.com/deviceplug/btleplug/blob/master/src/winrtble/ble/watcher.rs) : 
```rust
pub fn start(&self, on_received: AdvertismentEventHandler) -> Result<()> {
        self.watcher
            .set_scanning_mode(BluetoothLEScanningMode::Active)
            .unwrap();
        let handler = TypedEventHandler::new(
            move |_sender, args: *mut BluetoothLEAdvertisementReceivedEventArgs| {
                let args = unsafe { &*args };
                on_received(args);
                Ok(())
            },
        );
        self.watcher.add_received(&handler).unwrap();
        self.watcher.start().unwrap();  // <-- THERE
        Ok(())
}
```

**Expected behavior**
The error should be returned and not brutally unwrapped so I can handle it gracefully.

**Actual behavior**
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: HRESULT 0x800710DF', [~\.cargo\registry\src\github.com-1ecc6299db9ec823\btleplug-0.5.4\src\winrtble\ble\watcher.rs:48:3](https://github.com/deviceplug/btleplug/blob/master/src/winrtble/ble/watcher.rs#L48)

**Additional context**
I'm on Windows 10, and my bluetooth adapter is disabled in the command center.
[image]
Propagating this error allows me to handle this case gracefully.

qdot, 2021-02-05:
Ok, I *think* this is handled now. If not, please reopen or file something else.

qdot, 2021-02-05:
(Oh and by now I mean v0.6.0)

closed 2021-02-05 by qdot