deviceplug/btleplug #97: Allow to set custom timeout for connect

MattesWhite, 2020-12-02:
**Feature Description**

_Disclaimer:_ I can only speak for the Linux version. I don't know how this is handled for iOS or Windows.

At the moment, the timeout for a connect action is hard coded into the library: https://github.com/deviceplug/btleplug/blob/master/src/bluez/adapter/peripheral.rs#L404 
It would be nice to have a `connect_with_timeout(&self, timeout: Duration)` method to be able to adjust the timeout at runtime.

qdot, 2021-02-05:
Agreed, and this is still an issue in the new dbus bluez stuff. Will take this into consideration for the new API (cc @qwandor)

blandger, 2021-09-25:
As I can see that feature need enhancing bluez-async code.
I added that PR in bluez-async /first.
https://github.com/bluez-rs/bluez-async/pull/10

Added PR in btleplug
https://github.com/deviceplug/btleplug/pull/213/files

Massimiliano-solutiontech, 2023-08-15:
Currently the timeout is infinite? Because if that I'm gonna need this feature

blandger, 2023-08-15:
Not finished for Windows, macos.

https://github.com/deviceplug/btleplug/pull/213