deviceplug/btleplug #100: Add another maintainer for btleplug

MattesWhite, 2020-12-10:
@qdot I didn't found an e-mail so I'm writing this issue.

I see you are very active in developing buttplug-rs and I'm really impressed of how much you do. However, it seems to me (and I can understand that) that you lately spent more time on the main project as on this BLE crate. At the time of writing there are several open issues and a number of PRs. This shows that there is an interest in the Rust community to have a working cross-platform BLE crate. It's sad to see that you don't respond to the effort people spent to help (even if you probably can't due to time constraints).

To prevent btleplug from dying I propose you add another maintainer to the repository that can help you. At least with managing bug fixes. I know it is hard to give up some control but you shouldn't overwork yourself. I think with clear communication of your vision for btleplug (you wanted to switch to `async/.await`?) such a corporation can work and in the end buttplug.io will also benefit.

I don't want to apply as maintainer myself. As a PhD student I don't have much time either, but I'm sure you can find some volunteers either here or with a post on [TWiRs Call for Participation](https://users.rust-lang.org/t/twir-call-for-participation/). Well, maybe I could spent some hours managing PRs...

To emphasize again, I don't want to criticize your work. A cross-platform BLE crate is an amazing thing and the work done so far is good. I just hope that btleplug won't die because of a busy  maintainer.

Cheers MattesWhite

qdot, 2020-12-11:
Let me be clear: I don't want to maintain a bluetooth library.

I've worked on cross platform bluetooth in C++, C#, and Node. It's just not fun. I'm probably never going to need characteristic descriptors, service discovery on advertisement, and other things in this god forsaken mess of a communications protocol.

My problem is that I *need* a cross-platform bluetooth library, that works at least somewhat well on all platforms, and I haven't found one in any language that either stays maintained, or covers all the platforms I need. The only reason btleplug happened is because I saw the PRs in Rumble and knew where to cobble together other requirements to cover desktop to the very basic point that I needed, and I thought I'd at least share that much. That may have been a mistake.

In terms of bringing in other maintainers, I'm not real sure what you expect that to do here other than create more chaos. Most of the PRs fix one platform and break others (or in the case of macOS, have so far just broken everything, as my mac users are yelling at me about now), fix a problem someone was having in their own project that may not reflect the needs of a bluetooth API in general, or just straight up say "Didn't test on [x] platform because don't have it". That's why I've been stalling on bringing much in until an actual design document can be written, otherwise this will continue to be a shambling mass of half solutions with no guiding direction. The original API was myopically bluez focused, and really needs a complete rework, so I just don't have the energy to try and fix what's here currently because I honestly, truly do not think it is fixable. 

That said, it does what at least some people need for now in a minimal way, and in a way that no one else in Rust land has pulled off. This isn't good, it's just least bad.

If someone *does* think this is fixable in its current form, by all means, fork. I would be ecstatic to be proved wrong and use someone else's library. And if someone thinks they can bring in PRs here in a way that works across all platforms, I'd be happy to talk, but it's going to take a lot of time, a lot of concentration, and a lot of cross platform Bluetooth API knowledge that I'm not sure many people have.

And no, I don't have time to mentor.

MattesWhite, 2020-12-11:
I'm very very thankful for your long and honest answer.

Thank you for sharing btleplug and I can totally understand that it's a mess to work on a cross-platform BLE library. Me and my team already noticed a lot of points where other _mature_ BLE implementations are different in behaviour. I can clearly see that managing PRs for this is problematic.

It was never clear to me that the state of btleplug was enough for you. I always thought you would work on an async/.await API in secret and add further features later. Obviously this is not the case. Changing the README and adding a CONTRIBUTIONS file would show people better what they are dealing with. At the moment the readme suggests that there is still a lot of work in progress.

For now I have to discuss with my fellow team members what this means for our project.

qdot, 2020-12-11:
> It was never clear to me that the state of btleplug was enough for you. I always thought you would work on an async/.await API in secret and add further features later. Obviously this is not the [...]

I think my reply may have required some reading between the lines.

I don't *want* to do it, but afaict, no one else is going to either, so I'm kinda stuck. However, what the library does *right now* works for me in the most basic of ways, so I'm trying to live with that until the layers I need to finish above it are in a place they need to be. I would still very much like to circle back and fix things up. I would like even more if someone else did that for me, but unless they're doing it in a meticulous way for all platforms I need (which is pretty much "all of them"), I'm once again stuck.

There's work happening above this that pertains to this library in some ways, but may not be visible unless you're watching everything I do and trying to read the tea leaves there. For instance, I contributed a WebBluetooth IDL to web-sys in wasm-bindgen, and have a fully working Rust WASM WebBluetooth implementation in buttplug-wasm now. At some point, this will be ported down to btleplug so I can centralize all BTLE communication to the same library. Unfortunately, with the API the way it is right now, it'd be difficult to shoehorn it in that way.

The FFI implementation of buttplug-rs is also a trial run for figuring out how mobile FFIs will work here. My ultimate goal with btleplug is to also have it handling Android and iOS. Servo proved Android was possible, but the JNI is kind of a nightmare. iOS *may* already work with the CoreBluetooth impl, but I'd like to test that out on something easier first, then move down here.

So yeah, it's a weird situation, and this certainly isn't me floundering or refusing to accept help. However, this also isn't my first rodeo. I was the device interfaces WebAPI lead on Firefox during my time at Mozilla, I've done years of work outside of that on cross platform userspace device drivers, etc... This is a specialized part of software that not a ton of people give a shit about, and as I said, I unfortunately don't have time at the moment to teach anyone who might want to learn. Most people want it to "just work", and that's fair.

But with that, you're going to get what you pay for the price of admission at the moment, and that may mean slow updates. I'm not sure I want to go the same direction with this as I do Buttplug (see my discussion on this at https://twitter.com/qDot/status/1337094989621256194), but I'm also not going to lie and say that I care about anyone's project but mine, much as most contributors are making contributions to fix their specific project too.

closed 2021-01-26 by MattesWhite