deviceplug/btleplug #106: prompt NSBluetoothAlwaysUsageDescription in macOS Big Sur

chenpengfei, 2021-01-13:
[access] This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

qdot, 2021-02-05:
I'm gonna try to catch this soon. I just need a big sur machine to test on first.

qdot, 2021-04-29:
~3 months later, now installing Big Sur. Will see if I can make any headway on this.

Dajamante, 2021-07-01:
In my case, after running in lldb and getting the same error:
```
This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.
Process 91812 stopped
```
The solution was to grant "Full Disk Access" and "Bluetooth" to Terminal and ITerm in security settings. 
I am on macos Big Sur 11.2.2.

qwandor, 2021-07-01:
Thanks @Dajamante! That works for me too. We should add something in the documentation somewhere I guess.

qdot, 2021-07-31:
I... guess this is closable? We still don't have a good way to compile this into the binary (despite best efforts), but the Info.plist solution seems to work well enough and we've documented it, so calling this closed? Feel free to reopen or refile if you disagree.

closed 2021-07-31 by qdot

Dajamante, 2021-07-31:
I guess it is closable! Was able to work with btleplug, like a charm!

qdot, 2021-08-01:
> I guess it is closable! Was able to work with btleplug, like a charm!

Awesome! Ideally we'd be able to compile this permission in with the library, but I'm honestly not sure what proper etiquette is for that around the macOS ecosystem, so I think just making sure we provide info on how to do this at the binary level will hopefully be good enough. :)

PR #168 (2021-07-01): Check Bluetooth authorisation status on macOS, and add note about permissions
Fixes #106.