deviceplug/btleplug #111: What do we want to do with Write vs. WriteWithResponse?

qdot, 2021-02-04:
API bikeshedding time! Breaking out the discussion from #110 over here.

btleplug inherited its method names from rumble, which made some odd choices, I'm guessing either driven by bluez or the bluetooth spec. Request is basically WriteWithoutResponse, while Command I think is WriteWithResponse?

Anyways, in the new API, I think it'd be nice to collapse these to just write, as @qwandor mentioned. I think we *do* need to support both w/ and w/o response, possibly via a boolean argument, as multiple platforms can handle this capability (here's my old C# UWP code for handling that: https://github.com/buttplugio/buttplug-csharp/blob/master/Buttplug.Server.Managers.UWPBluetoothManager/UWPBluetoothDeviceInterface.cs#L296), and there are devices that expose both characteristic properties.

In my libraries that live on top of this, I have a full Write options object that can be built, but that's because I have to support Bluetooth/Serial/USB/god knows what else. This signature could probably just be `write(data: &[u8], write_with_response: bool)`? Or we could just have write/write_with_response methods (this might actually be clearer). If someone requests a type a characteristic doesn't handle, we throw an error.

qwandor, 2021-02-04:
I think it's the other way around, `request` is write, while `command` is write-without-response. But I agree that the current names are confusing.

I'm curious what the usecases are for explicitly choosing one over the other. If a characteristic supports write, when would you want to do `write-without-response` instead?

qdot, 2021-02-05:
> I'm curious what the usecases are for explicitly choosing one over the other. If a characteristic supports write, when would you want to do `write-without-response` instead?

Latency. Remember, most of my use case is devices that need to be control in a "as real time as possible" way. Is this what BLE was made for? Absolutely not. But since BLE is what modern phones support, and since we can't always access the BLE frame timings on all platforms (and BLE, by spec, *requires* transactions to run in order of queuing and complete successfully), we're kinda stuck.

Using `write` means that a GATT transaction is not finished until the response is received. This can take A While (10s-100s of milliseconds) in some cases, and on some platforms (WebBluetooth, for instance), we can't send another GATT transactions to the characteristic until the last one is finished. So sometimes, even when we have both capabilities on a characteristic, we just want a "fire and forget" style interface, which is what `write` gives us.

MattesWhite, 2021-02-05:
Another point to keep the distinction of WriteWithoutResponse/Command and Write/Request is that they are actually two different things.

Here is a blog post about it: https://iotexpert.com/ble-write-request-write-command-signed-write-command-prepare-write/

**tl;dr**

- A Command is limited to a single package, i.e. its contents must not exceed the MTU (maximal transport unit) and thus is limited.
- A Request will be acknowledged or an error response is sent by the GATT server. The error is encoded in one byte and values `0x01` through `0x07` are defined by the Bluetooth spec (see the post). Other error values can be customly assigned. In the end, I think this can be handled by simply adapt the variants of the error returned by a Request.

Therefore, having a `write`/`request` and a `write_without_response`/`command` method is required. However, in many use-cases a characteristic only allows one of the write methods so a more general method that picks what to do based on the characteristic's properties would be nice.

qwandor, 2021-02-05:
Right, they are different things at the protocol level, but that doesn't necessarily mean they need to be at the API level. The point about latency is good though. Comparing the system APIs:

 * BlueZ [`GattCharacteristic1.WriteValue`](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt) has an option to specify the type of write, with three possible values: write with response, write without response, or reliable write. If it's not specified, like at present, I assume it will default to whatever the characteristic supports.
 * CoreBluetooth [`writeValue:for:type`](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518747-writevalue) has a mandatory parameter to specify either write with response or write without response.
 * I'm not very familiar with the Windows API, but it looks like there's an optional [`GattWriteOption`](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattwriteoption?view=winrt-19041) enum which allows to specify write with response or write without response.

So it should be possible to do something similar on all three platforms. Options are:
 * Two separate methods, as we currently have. I'd prefer to call them `write` and `write_without_response` to match the characteristic flags, though this is a change from the current names.
 * One method with an enum flag to choose what kind of write to do. This enum could perhaps have a third value 'automatic' or something like that, to just do whatever the device supports, in the common case that the developer doesn't really care about the distinction. It's not clear what this automatic option would do on Mac OS though, where the parameter is mandatory.

qdot, 2021-02-05:
Yup, the `GattWriteOption` in UWP would basically be similar to the enum here.

I'm a fan of "say what you mean" here so I'd rather avoid the automatic option. I've not run into a BLE API so far that doesn't make the distinction at all, and in cases where there are both write and write_with_response characteristics we should make the developer make that decision versus making it for them.

As for method versus enum question, seems like all the platforms handle this with an enum tacked on to the write method parameters, so I'd be fine with that too.

qdot, 2021-02-23:
Fixed in #113

PR #113 (2021-02-14): Replace request and command with a single write method.
This is hopefully easier to understand, and more similar to other GATT APIs.

It appears that the write types were backwards on Mac OS before, so this also fixes that.

I don't have a Windows machine to build on, so I haven't been able to implement this on Windows.

Fixes #111.
closed 2021-02-23 by qdot