deviceplug/btleplug #119: Panic via Uuid::from_str() (Corebluetooth, notifications)

darrinsmart, 2021-02-14:
I'm experimenting with btleplug to read values from a smart scale.

My code is subscribing to the characteristic 00:00:FF:E4:00:00:10:00:80:00:00:80:5F:9B:34:FB - aka FF:E4 by calling Peripheral::subscribe(). That part seems to be fine. But after the subscription is done, I see an error in a background thread:

```
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error(Parser(InvalidLength { expected: Any([36, 32]), found: 4 }))', /Users/darrin/src/btleplug/src/corebluetooth/central_delegate.rs:479:86
stack backtrace:
   0: rust_begin_unwind
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:493:5
   1: core::panicking::panic_fmt
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/panicking.rs:92:14
   2: core::option::expect_none_failed
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/option.rs:1268:5
   3: core::result::Result<T,E>::unwrap
             at /Users/darrin/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/src/rust/library/core/src/result.rs:973:23
   4: btleplug::corebluetooth::central_delegate::CentralDelegate::delegate_peripheral_didupdatenotificationstateforcharacteristic_error
             at /Users/darrin/src/btleplug/src/corebluetooth/central_delegate.rs:479:21
```

It looks like the error is this line in delegate_peripheral_didupdatenotificationstateforcharacteristic_error():
```
        let cuuid = Uuid::from_str(&NSStringUtils::string_to_string(cuuid_nsstring)).unwrap();
```
cuuid_nsstring is "FFE4", which uuid::Uuid::from_str() refuses to parse. Perhaps a fix would be to use the new UUID type from pull request #81?

qwandor, 2021-02-14:
Thanks for the bug report!

I don't think #81 fixes this issue, as it doesn't change the conversion from `NSUUID` to `Uuid`. I've made a PR #120 which I think should, can you please try patching that in and see if it works for your code? You should be able to do this by changing the btleplug dependency in your `Cargo.toml` to

```
btleplug = { git = "https://github.com/qwandor/btleplug", branch = "uuid" }
```

darrinsmart, 2021-02-14:
Thanks for the fast response! And the Cargo.toml tip as well.

Your PR looks good, I am now able to subscribe & get a notification message from my device. 

Thanks again.

NZSmartie, 2021-03-02:
Closing as #120 is merged and is released in `v0.7.0`

PR #120 (2021-02-14): Add helper function to convert CBUUID to Uuid.
This fixes a bug where 16-bit UUIDs were not being converted properly.

Fixes #119.
closed 2021-03-02 by NZSmartie