deviceplug/btleplug #122: 0.6.x passive bluetooth scan support

braincow, 2021-02-16:
**Implement passive scans**

Currently 0.6.0 gives error when trying to activate passive scanning on the interface. Passive scans are useful for listening Bluetooth beacons in LTE environments.

qwandor, 2021-02-16:
Thanks for the bug report! Some more details would be helpful. In particular:
 * What platform are you having this problem on? Does it work on other platforms?
 * What error are you getting?
 * Can you provide some sample code to reproduce the issue?
 * Is this a new issue in 0.6.0 that worked in earlier versions?
 * Does it work on the dev branch?

MattesWhite, 2021-02-19:
I have the same problem working on Linux.

The problem I see is that active/passive scan (as well as filter duplicats option) is no longer implemented in the new `dbus` version (only a `todo!()` macro).

see https://github.com/deviceplug/btleplug/blob/master/src/bluez/adapter/mod.rs#L515

qwandor, 2021-02-19:
The [D-Bus interface of BlueZ](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt) does not allow us to choose between active and passive scanning, unfortunately. But active scanning should always give a superset of the information that passive scanning gives, so I'm not sure what the issue is other than power usage?

The filter duplicates option will be implemented after the async API change.

braincow, 2021-02-24:
> The [D-Bus interface of BlueZ](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt) does not allow us to choose between active and passive scanning, unfortunately. But active [...]

Power usage for one. Second, some hardware seems to have a timeout set for the duration of the active scan. Therefore one needs to continually be checking the state of the interface and start the scanning again if it stops in code. Passive scanning to my understanding has no such timeout (at least on my hardware I have available). This is an annoyance at best, show stopper at worst, on some lower power devices when just trying to record continuously incoming Bluetooth beacon data.

NZSmartie, 2021-02-24:
This is mostly my fault for not investigating the passive scanning further and leaving behind a `todo!()` during #105 
From looking at the BlueZ source code, it does perform passive and active scanning, however it switches between them indirectly (there's no direct API for it as @qwandor) has already stated, and requires fiddling around with how the `start_discovery` amongst other things. I'm in the middle of some other changes, but will give this a look at again soon.

Honestly, this is a bad design decision on BlueZ's part

braincow, 2021-02-25:
> 
> Honestly, this is a bad design decision on BlueZ's part

I am inclined to concur on you with this. :)

However, I am unable to use 0.6.x until there is a way to enable passive scanning and will stick with older version for now. Keep us posted if there is a change so that we can test. Thanks in advance.

qdot, 2021-07-31:
There's also no way to select active/passive scanning on most of the non-Bluez platforms we support (and from the sound of this, our BlueZ impl will have issues), and this library is breadth (platforms) versus depth (every API possible) first, so calling this invalid and closing.

closed 2021-07-31 by qdot