deviceplug/btleplug #132: Link to documentation

piegamesde, 2021-03-03:
There are hosted docs for this crate [on docs.rs](https://docs.rs/btleplug/0.7.1/btleplug/), but neither the README nor the crates.io home page link to them for some reason.

qwandor, 2021-03-03:
crates.io does have a link in the usual place, but adding it to the README is a good idea.

piegamesde, 2021-03-03:
> crates.io does have a link in the usual place

Weird. It doesn't for me
[image]

qwandor, 2021-03-03:
Huh, that is indeed weird. It shows it for me on the right, between 'Install' and 'Repository', whether or not I'm logged in. Does it work for other crates?

piegamesde, 2021-03-03:
> Does it work for other crates?

On some yes, on other not. https://crates.io/crates/ash has a doc link for me, https://crates.io/crates/blurz not.

qdot, 2021-07-31:
Docs link is in our README now, closing.

PR #133 (2021-03-04): Add docs.rs badge with link to latest published docs.
Fixes #132.
closed 2021-07-31 by qdot