deviceplug/btleplug #138: Ability to fetch radio info

qdot, 2021-03-18:
**Feature Description**

This may suck to implement cross-platform, but it'd be nice to have a way to get what brand/model of bluetooth radio is being used by the library. There are *many* times where I need to ask users what radio they're using when they have issues, and having a way to fetch that from the bluetooth library could automate that step.

I have some C# code for UWP where I was getting this on windows:

https://github.com/buttplugio/buttplug-csharp/blob/master/Buttplug.Server.Managers.UWPBluetoothManager/UWPBluetoothManager.cs#L81

I suspect there's something similar available for Bluez. I suspect that this would really only be needed for pre-2012 macOS machines (before bluetooth 4 radios were standard), so we could probably just return something default there for now.

qwandor, 2021-08-30:
With BlueZ you can get either the device ID (e.g. "hci0") or the modalias (which contains the USB vendor and product ID, if it's a USB adapter). I guess you'd want the latter for debugging, but we could expose both.

qdot, 2021-08-30:
Honestly anything helps. As usual, my concerns are far more about windows than anything, but if we can get it on Linux also, great (macOS we can mostly assume it's built in, unless they're on a pre-2012 machine). I used to be able to do Management API dumps on the windows side, I may look at that again.