deviceplug/btleplug #145: Version 0.4 fails trying compiling socket2 v0.4.0

temphiza, 2021-04-05:
**Describe the bug**
Using version 0.4 is failing trying to compile socket2 v0.4.0

**Expected behavior**
Normal cargo installation

**Actual behavior**
Package btleplug not installed

**Additional context**
error[E0658]: `match` is not allowed in a `const fn`
   --> ../.cargo/registry/src/github.com-1ecc6299db9ec823/socket2-0.4.0/src/lib.rs:156:9
    |
156 | /         match address {
157 | |             SocketAddr::V4(_) => Domain::IPV4,
158 | |             SocketAddr::V6(_) => Domain::IPV6,
159 | |         }
    | |_________^

qdot, 2021-04-06:
We're on version 0.7.2 now? Is there a reason you need version 0.4?

temphiza, 2021-04-06:
I was using dependency string from README:

btleplug = { version = "0.4", features = ["serde"] }

I just switch to version 0.7.2 and having the following error:

error[E0277]: the trait bound `Vec<u8>: Serialize` is not satisfied
   --> .../.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.7.2/src/api/mod.rs:334:9
    |
334 |         data: Vec<u8>,
    |         ^^^^ the trait `Serialize` is not implemented for `Vec<u8>`
    |
    = note: required by `serde::ser::SerializeStructVariant::serialize_field`

The dependencies I have:

[dependencies]
chrono = "0.4"
btleplug = { version = "0.7.2", features = ["serde"] }

I developing on macos 11.2.3 with rustc 1.51.0,

Thanks four your help!

temphiza, 2021-04-06:
I just change dependency value to: btleplug = "0.7.2" and it works!

Thanks again for your help!

closed 2021-04-06 by temphiza