deviceplug/btleplug #147: Calling manager.adapters() results in process abort on macOS

Mallets, 2021-04-14:
**Describe the bug**
As per title, calling manager.adapters() results in process abort on MacOS.

**Expected behavior**
The process shouldn't abort.

**Actual behavior**
As per title, calling manager.adapters() results in process abort on MacOS.

**Additional context**
Please find below the info about my system:

- macOS Big Sur - Version 11.2.3
- btleplug from master (commit [33e5acd06d17235bb932fa33e3fb6973c57905bd](https://github.com/deviceplug/btleplug/commit/33e5acd06d17235bb932fa33e3fb6973c57905bd))
- Rust stable - rustc 1.51.0 (2fd73fabe 2021-03-23)

The code that generates the process abort is as simple as below. If I comment the instruction `let _ = manager.adapters();`, the program exists without aborting as expected.
```
#[cfg(target_os = "linux")]
use btleplug::bluez::manager::Manager;
#[cfg(target_os = "macos")]
use btleplug::corebluetooth::manager::Manager;
#[cfg(target_os = "windows")]
use btleplug::winrtble::manager::Manager;

fn main() {
    let manager = Manager::new().unwrap();
    let _ = manager.adapters();
}
```

The output is:
```
$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.92s
     Running `target/debug/baseline`
[1]    17100 abort      cargo run
```

Trying to debug it with LLDB, I get the following:
```
$ lldb cargo run
(lldb) target create "cargo"
Current executable set to 'cargo' (x86_64).
(lldb) settings set -- target.run-args  "run"
(lldb) run
Process 17129 launched: '/Users/luca/.cargo/bin/cargo' (x86_64)
Process 17129 stopped
* thread #2, stop reason = exec
    frame #0: 0x0000000100f41000 dyld`_dyld_start
dyld`_dyld_start:
->  0x100f41000 <+0>: popq   %rdi
    0x100f41001 <+1>: pushq  $0x0
    0x100f41003 <+3>: movq   %rsp, %rbp
    0x100f41006 <+6>: andq   $-0x10, %rsp
Target 0: (cargo) stopped.
(lldb) thread backtrace
* thread #2, stop reason = exec
  * frame #0: 0x0000000100f41000 dyld`_dyld_start
(lldb)
```

I'm not 100% sure how to read this, it seems related to some dynamic library being loaded at startup.
Any help or hint on how to investigate the problem further would be very much appreciated!

qwandor, 2021-04-14:
This might be related to #106, but your error looks different to what I was seeing. Can you try:

1. Using the `dev` branch of btleplug rather than `master`.
2. Running the binary you built under lldb rather than cargo, i.e. `cargo build` then `lldb target/debug/baseline`.

Mallets, 2021-04-14:
Thanks for the quick response.

1. It doesn't compile with the `dev` branch.
```
error[E0603]: module `corebluetooth` is private
   --> src/main.rs:4:15
    |
4   | use btleplug::corebluetooth::manager::Manager;
    |               ^^^^^^^^^^^^^ private module
    |
note: the module `corebluetooth` is defined here
   --> /Users/luca/.cargo/git/checkouts/btleplug-4193de4d1cdc37dd/b024c36/src/lib.rs:102:1
    |
102 | mod corebluetooth;
    | ^^^^^^^^^^^^^^^^^^
```

2. Here the LLDB output with `master` branch.
```
$ lldb ./target/debug/baseline
(lldb) target create "./target/debug/baseline"
Current executable set to '/Users/luca/Repository/rust-net/bluetooth/baseline/target/debug/baseline' (x86_64).
(lldb) run
Process 19887 launched: '/Users/luca/Repository/rust-net/bluetooth/baseline/target/debug/baseline' (x86_64)
2021-04-14 11:47:04.805675+0200 baseline[19887:203579] [access] This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.
Process 19887 stopped
* thread #12, queue = 'com.apple.root.default-qos', stop reason = signal SIGABRT
    frame #0: 0x00007fff20431792 libsystem_kernel.dylib`__abort_with_payload + 10
libsystem_kernel.dylib`__abort_with_payload:
->  0x7fff20431792 <+10>: jae    0x7fff2043179c            ; <+20>
    0x7fff20431794 <+12>: movq   %rax, %rdi
    0x7fff20431797 <+15>: jmp    0x7fff204106a1            ; cerror_nocancel
    0x7fff2043179c <+20>: retq
Target 0: (baseline) stopped.
(lldb) thread backtrace
* thread #12, queue = 'com.apple.root.default-qos', stop reason = signal SIGABRT
  * frame #0: 0x00007fff20431792 libsystem_kernel.dylib`__abort_with_payload + 10
    frame #1: 0x00007fff204331d9 libsystem_kernel.dylib`abort_with_payload_wrapper_internal + 80
    frame #2: 0x00007fff2043320b libsystem_kernel.dylib`abort_with_payload + 9
    frame #3: 0x00007fff24ca71b4 TCC`__TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163
    frame #4: 0x00007fff24ca79e6 TCC`__TCCAccessRequest_block_invoke.128 + 498
    frame #5: 0x00007fff24ca560a TCC`__tccd_send_message_block_invoke + 573
    frame #6: 0x00007fff20188ab8 libxpc.dylib`_xpc_connection_reply_callout + 36
    frame #7: 0x00007fff20188a40 libxpc.dylib`_xpc_connection_call_reply_async + 69
    frame #8: 0x00007fff20299831 libdispatch.dylib`_dispatch_client_callout3 + 8
    frame #9: 0x00007fff202b2010 libdispatch.dylib`_dispatch_mach_msg_async_reply_invoke + 375
    frame #10: 0x00007fff202a9564 libdispatch.dylib`_dispatch_kevent_worker_thread + 1390
    frame #11: 0x00007fff204414e1 libsystem_pthread.dylib`_pthread_wqthread + 386
    frame #12: 0x00007fff20440467 libsystem_pthread.dylib`start_wqthread + 15
(lldb)
```

qwandor, 2021-04-14:
Yep, that's #106. Closing as a duplicate.

(For compiling on the `dev` branch, you can remove all the `#[cfg]` stuff and just `use btleplug::platform::{Adapter, Manager, Peripheral};`.)

closed 2021-04-14 by qwandor