deviceplug/btleplug #150: bluez scan runs into DBUS limits

Wilm0r, 2021-04-18:
**Describe the bug**
By subscribing to every device during a scan, `btleplug` runs into DBUS subscription limits pretty quickly, then stops working.

**Expected behavior**
An ability to scan without permanently subscribing to every device ever seen.

**Actual behavior**
After a few hours or days, depending on the limit set in DBUS systemwide configs, you'll get something like this in your syslogs:

`Apr 17 00:02:01 radar dbus-daemon[285]: [system] Connection ":1.42208" is not allowed to add more match rules (increase limits in configuration file if required; max_match_rules_per_connection=20480)`

**Additional context**
I've written a Prometheus exporter for my BLE temperature sensors using btleplug. It works great until DBUS kills the library. I then have to wait for some time, restarting immediately doesn't work because the list of peripherals sticks around in DBUS' memory, tricking btleplug into subscribing to them all again right away after restart.

For my BLE scan I obviously only need info from the BLE sensors, it would be great if I could opt out from unsupported devices, or do something else to avoid hitting DBUS limits.

(Apologies for my likely poor BLE/DBUS terminology here.

qwandor, 2021-04-18:
What version of btleplug are you seeing this issue in? Can you try running against the latest `dev` branch and see if you still have the same issue there? There have been a lot of changes since the last release, especially to the Linux implementation. (It's also moved to async, so you'll need to update your code a bit to match.)

Wilm0r, 2021-04-20:
This was using a crated version, indeed not dev yet.

So thanks for the suggestion! Was keen to move to async anyway. I've updated my exporter and first experiments seem promising; I've lowered the dbus limit again but was able to immediately restart using the dev binary.

Let's observe for a few days, but I guess this bug can be closed, yes. :-)

Wilm0r, 2021-04-26:
It's been rock solid for days now, thank you!

Any idea on when the current dev code will hit crates.io?

qwandor, 2021-04-26:
@qdot, thoughts?

qdot, 2021-04-26:
@qwandor Ok, so we've gotten some dogfooding happening on Linux, which is great. Windows is still waiting on me to port but I think I have temporarily escaped Unity Hell for now so I can maybe look at that... Do we have anyone looking at macOS much, or are we punting on that until the permissions bug gets worked out?

qdot, 2021-07-31:
v0.8 is out now (wow. 3 months between commenting here and getting that done. Go me. :( ), closing.

closed 2021-07-31 by qdot