deviceplug/btleplug #155: Windows reads Gatt Characteristic from Windows Cache by default

jsosulski, 2021-05-02:
**Describe the bug**
I have a bluetoothLE temperature sensor. When reading a Gatt Characteristic from windows, the first value read after a sensor connect is always the same value until I disonnect / reconnect the device.

**Expected behavior**
Running the same program under Linux I get changing temperature values with each read, without needing to reconnect the device everytime.

**Actual behavior**
See above.

**Additional context**
I think windows per default always reads from the Windows Cache if it is available. Direct quote from the [documentation](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.readvalueasync?view=winrt-19041):

`ReadValueAsync():     Performs a Characteristic Value read from the value cache maintained by Windows.`

I do have a fix (see the [diff here](https://github.com/deviceplug/btleplug/compare/dev...jsosulski:win_force_read_from_device?expand=1).) that solves the problem but unfortunately I am not aware of any side effects and whether you would want to expose the option whether to read from cache or not. However, this would unify behaviour between linux and windows for now.

NZSmartie, 2021-05-02:
This is a good catch, I strongly encourage you to open a pull request with the fix unless you'd like me to go ahead and do that for you.
As there is no API to control the caching behavior in btleplug, this is a good fix in my opinion. 

Personally, I encountered this issue not too long ago, but did not investigate further.

jsosulski, 2021-05-02:
I opened a PR.

One more note though: apparently the Cache should become invalidated ['whenever the remote device indicates a service change'](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothcachemode?view=winrt-19041#remarks).

I dont know why that does not happen, but it could also be that my sensor is not sending the stuff it should send :)

qwandor, 2021-05-02:
Are you sure the characteristic is just a read characteristic and not notify or something? Most sensors you need to subscribe to notifications.

jsosulski, 2021-05-02:
My sensor offers both actually, one characteristic to subscribe and another one that contains the current temperature.

For now I went the less elegant route of periodically reading out the characteristic that has the temperature stored. Anyways, I just noticed that this worked out-of-the-box when compiled on Linux and needed the change in PR #156 when compiled on Windows.

sergchan, 2021-07-02:
Hi,
I detected it on my PC and my ble-server. I written ble-client on python for check my server and discovered that it is relevant for me. At every reading I receive same package when use Rust and btleplug.

qwandor, 2021-07-02:
@sergchan Can you try on the latest dev branch, if you're not already? The bug should be fixed there.

sergchan, 2021-07-05:
@qwandor HI,
Thanks for the quick response.
I found sample code (branch: dev) for async fn main.
Is sync code still supported?

sergchan, 2021-07-05:
I also found that .connect() not worcking.

qwandor, 2021-07-05:
The next release of btleplug will be async. If you want to continue to use it from sync code, you can use an executor which blocks on each call.

qdot, 2021-07-31:
Pretty sure this got fixed in v0.8 (the cache mode is now explicitly set), closing.

PR #156 (2021-05-02): Force  to read from GattCharacteristic from device instead of cache
Closes #155 

For some reason `BluetoothCacheMode(1)` is the no caching mode and `BluetoothCacheMode(0)` uses the cache. See [docs](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothcachemode?view=winrt-19041#fields)

Using `BluetoothCacheMode::Uncached` did not work for me. Maybe this is not really an enum? My IDE was also not very helpful regarding these windows bindings.
closed 2021-07-31 by qdot