deviceplug/btleplug #165: (dev branch - async) Notifications not firing correctly on Linux Bluez

qdot, 2021-06-05:
**Describe the bug**
1. Connect to a device with a notify endpoint
2. Subscribe to endpoint notifications
3. Trigger notification

**Expected behavior**
Notification event is fired

**Actual behavior**
We're at least getting PropertiesChanged updates, as I'm seeing this in the logs:
```
Jun 05 23:50:36.502 TRACE bluez_async::events: PropertiesChanged for /org/bluez/hci0/dev_C1_E0_E9_13_05_7C/service000b/char000e: PropertiesPropertiesChanged { interface_name: "org.bluez.GattCharacteristic1", changed_properties: {"Value": Variant([80, 58, 50, 48, 52, 58, 67, 49, 69, 48, 69, 57, 49, 51, 48, 53, 55, 67, 59])}, invalidated_properties: [] }
```

However, we're not getting any events sent back through the streams.

**Additional context**

This test works on Win/macOS.

qdot, 2021-06-21:
If I watch the main event_stream (which gets all events off the central), we do get characteristic updates in btleplug there. However, using the device_event_stream filter, we get nothing at the device level.

@qwandor this might be an issue back in bluez-async? Do you have any devices with notify/indicate characteristics to test with?

qwandor, 2021-07-03:
I just tried to reproduce this, but it seems to work fine for me. Using the notification example from #157 (but with a different device), I'm able to subscribe to notifications and get them as expected on Linux. I tried several times and it seemed to work fairly reliably too.

qdot, 2021-07-03:
@qwandor Ok, thanks for trying. This could very well be something weird up in Buttplug, so I'll adapt the notification example to my needs and see if I can repro.

blandger, 2021-07-15:
I've used that https://github.com/deviceplug/btleplug/pull/157 with one of 'old neuro device' and it works fine.
But when I use the same code for 'newer neuro device' (using updated UUIDs/name) that example doesn't work for me too on Linux.
I will try one work around with not working device. may be that is kind of 'feature' of hardware. I'll write my research result then.

blandger, 2021-07-15:
Ok, I tried work around for mi ble device.
I found that approach in C++ code for device: _from silicanlabs forum - write 0x0001 to enable notifications_
So I tried to write data first as mentioned
```
       if characteristic.uuid == WRITE_TO_CHARACTERISTIC_UUID
              && characteristic.properties.contains(CharPropFlags::WRITE)
       {
           let data:[u8;2] = [00, 01]; // I tried  [01, 00] too
           println!("Sending command {:?} to characteristic {:?}", &data, characteristic.uuid);
           peripheral.write(&characteristic, &data, WriteType::WithResponse).await?;
        }
      // Subscribe to notifications from the characteristic with the selected UUID.
        if characteristic.uuid == NOTIFY_CHARACTERISTIC_UUID
               && characteristic.properties.contains(CharPropFlags::NOTIFY)
        {
           println!("Subscribing to characteristic {:?}", characteristic.uuid);
```

but that didn't help. Example stucks on _Subscribing to characteristic xxxx_ message and hangs at that point.

blandger, 2021-07-15:
May be there is something odd with subscribing on notification. I would like to help track the problem if library has ability to write full internal logs while running example with my device on linux.

qwandor, 2021-07-15:
Try adding `pretty_env_logger::init()` to the start of the main function (we should have added this in the first place), and then set `RUST_LOG=trace` and you should get more logs.

blandger, 2021-07-16:
[trace.log](https://github.com/deviceplug/btleplug/files/6831892/trace.log)
Here is a trace log for example.

blandger, 2021-07-18:
[working_trace.log](https://github.com/deviceplug/btleplug/files/6836271/working_trace.log)
Here is a trace for similar device (but older hardware) that works fine with example code.

blandger, 2021-07-19:
As I can assume, two logs are not quite verbose and informative at interested code part, aren't they? If someone prepares different branch with more correct logging, I will run both examples again to gather data.

qdot, 2021-07-24:
@qwandor Just updated this with the relevant info that we were discussing on discord.

dnseitz, 2021-08-05:
Is there any update on this? I'm running into the same issue.

qdot, 2021-08-05:
> 
> 
> Is there any update on this? I'm running into the same issue.

Nope. Right now I'm just using the "event detect on mac/win, polling on linux" workaround. :(

qwandor, 2021-08-05:
I can try to look at this, but it won't be for a few weeks probably. Does someone have a minimal example to reproduce it yet? Last time I looked I wasn't able to reproduce the issue.

qdot, 2021-08-05:
> I can try to look at this, but it won't be for a few weeks probably. Does someone have a minimal example to reproduce it yet? Last time I looked I wasn't able to reproduce the issue.

Did you try with the event discovery example? That does it for me with pretty much any device.

qwandor, 2021-09-06:
The underlying bug is https://github.com/diwic/dbus-rs/issues/346, which should be fixed by https://github.com/diwic/dbus-rs/pull/347.

blandger, 2021-09-25:
Just checked on my HW, it still doesn't work on btleplug=0.8.1 with dbus = "0.9.3"
I see some changes were merged into 0.9.3
https://github.com/diwic/dbus-rs/pull/351

Is there something I can try to fix on local dbus fork to check and probably make it work ? Where can I look at in dbus code?

qwandor, 2021-09-25:
I'm waiting for the next release of the `dbus` crate with the fix in it, then I'll update `bluez-async` and `btleplug` to use it. We need to call the new `set_all_signal_matches` method to enable the mode that works properly.

PR #214 (2021-10-02): Update to latest version of bluez-async
This should fix #165. Fixed some warnings while I was at it.