deviceplug/btleplug #172: `api::PeripheralProperties::local_name` is always `None` in macOS

hideakitai, 2021-07-24:
**Describe the bug**

There are two problems in this issue:

1. CoreBluetooth does not have the `local_name` data in the first discovery

This may be the problem of CoreBluetooth API, but there is no `local_name` data (NULL) inside of `peripheral` here:
https://github.com/deviceplug/btleplug/blob/33e5acd06d17235bb932fa33e3fb6973c57905bd/src/corebluetooth/central_delegate.rs#L357

But we can find the valid name if the advertisement data has it, after the first one.

2. `Peripheral` data is not updated after the first discovery

This may be related to #117. `local_name` can be found after the first discovery, but it's not reflected `PeripheralProperties` stored inside of the `Peripheral` struct.

So, stored `PeripheralProperties` inside of `Peripheral` struct is remained as the state of the first discovery: i.e. `local_name` is `None`.

**Expected behavior**

We can get `local_name` if the advertisement data has it.

**Actual behavior**

`local_name` is `None` even if its advertizement data has `local_name`.

qdot, 2021-07-24:
You might want to check out the dev branch (or possibly dev-updates, we're running a bit behind on getting everything merged right now), as we've got some pretty massive changes coming in soon (basically an overhaul of the API to move to async) and some of these things may be fixed there.

qdot, 2021-07-25:
Thanks for the patches!

I'm bringing this and your other PR into master and cutting 0.7.3, after which I'll be bringing in the dev branch and starting the 0.8.x series. This is going to be a bumpy ride though, so I expect some people may stick to 0.7.x while we make sure 0.8.x works, which is why I wanna get this patches in.

Also wow Rhizomatiks! Love y'alls work, I was at the Elevenplay show in SF pre-COVID. :D

hideakitai, 2021-07-26:
Thanks for the great crate. I'm still getting used to Rust (and async), so I was hoping to get 0.7.x working first, but of course I want to benefit from async, so I'll try the dev branch later :)
I'm really looking forward to the 0.8.x series!

Wow, thanks for coming to the show! I wasn't at the show in SF, but I'm sure the devices and systems were working fine. I hope we'll be able to tour around the world again soon.

closed 2021-07-26 by hideakitai