deviceplug/btleplug #176: Implement DeviceLost event on WinRT and BlueZ

qdot, 2021-07-31:
**Feature Description**

DeviceLost is currently only fired by macOS. BlueZ had it but lost it after @NZSmartie's patches got written over, and I'm not sure if it's even a thing on Windows so we might have to do our own timing there to emulate it.

qwandor, 2021-07-31:
It looks like the MacOS implementation is firing `DeviceLost` in response to `on_peripheral_disconnect`, which should be `DeviceDisconnected`. What is `DeviceLost` supposed to mean, actually?

qdot, 2021-07-31:
Oh. Well. That's a completely different bug then heh.

I just went back and looked at rumble, and they never implemented DeviceLost, just had it in their enum, with no explanation. So, gonna change this bug to "Remove DeviceLost and change macOS to DeviceDisconnected"

qwandor, 2021-08-25:
Fixed in dev.

PR #193 (2021-08-24): Remove CentralEvent::DeviceLost and rename CoreBluetoothEvent::DeviceLost
`CentralEvent::DeviceLost` was never constructed anywhere. `CoreBluetoothEvent::DeviceLost` actually corresponds to `CentralEvent::DeviceDisconnected`, so rename it to match.

Fixes #176.
closed 2021-08-25 by qwandor