deviceplug/btleplug #223: 0.9.0 crashes on OSX

coral, 2021-11-01:
After upgrading from 0.8.0, I am experiencing crashes when trying to connect to periphs on 0.9.0

```
thread '<unnamed>' panicked at 'We should still have a future at this point!', /Users/coral/.cargo/registry/src/github.com-1ecc6299db9ec823/btleplug-0.9.0/src/corebluetooth/internal.rs:201:17
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: BTLEError(Other("Channel closed"))', examples/listen.rs:15:51
```

The same code works when downgrading to 0.8.0 again

coral, 2021-11-01:
[image]

maybe the QUESTIONABLE ASSUMPTIONS are actually QUESTIONABLE :D

qwandor, 2021-12-29:
Thanks for the bug report.

Are you able to reproduce this with one of the examples, or some other simple and reliable reproduction?

coral, 2022-01-01:
@qwandor Was a while ago but i'm 99% certain that I tried with the example prior to filing this report. Once I'm back home i'll test this again to confirm.

pmnair01, 2022-03-21:
@qwandor I am working on a cross-platform utility and the code that works on Linux seems to hit this panic on mac os (Big Sur 11.6.4). The code is still in simple POC and here is what it does
* scan for device with filter (filter service UUID)
* connect to peripheral
* get device services
* READ/WRITE from any characteristic
The last is successful on Linux, however I see the panic on my mac. Any ideas as to what could be the reason?
@coral were you able to get this to work?

vvhh2002, 2022-05-26:
` DEBUG btleplug::corebluetooth::future  > Waker set.
thread '<unnamed>' panicked at 'We should still have a future at this point!', /btleplug-0.9.2/src/corebluetooth/internal.rs:205:17
 ERROR btleplug::corebluetooth::peripheral > Event receiver died, breaking out of corebluetooth device loop.`

I had the same issue

qdot, 2022-11-07:
May be fixed as part of #274 or our other fixes involving CoreBluetooth's weird futures issues.

closed 2022-11-07 by qdot

coral, 2022-11-15:
Can confirm this has solved my issues at least