deviceplug/btleplug #224: Bluetooth discovery doesn't work on macOS Monterey

gbaranski, 2021-11-02:
**Describe the bug**
Bluetooth discovery doesn't work on MacOS Monterey

**Expected behavior**
Scanning does discover devices.

**Actual behavior**
Scanning doesn't discover any devices. 

**Additional context**
Running on macOS Monterey. Tested using few terminal emulators, permissions are granted. It works on macOS Big Sur. Probably related to https://github.com/hbldh/bleak/issues/635 and https://stackoverflow.com/questions/69661809/no-name-or-address-cbcentralmanager-no-longer-working-on-macos-12

subtle-supernova, 2021-12-15:
I've confirmed this is still an issue in MacOS 12.1. I tried to pass a UUID to `ScanFilter` similar to what was outlined here: https://github.com/hbldh/bleak/commit/5e1a9b334f5a72dc7d59aa398230fa997dca1c40 with something like

```rust
let scan_filter = ScanFilter{services: vec!(uuid_from_u16(0xFDCE))};
```

Of course I could have done that way wrong as this is my first time doing anything with bluetooth in my life, but decided to add this here in case someone stumbles upon a working solution using `ScanFilter` or this is useful in some way.

qdot, 2021-12-15:
afaik we don't actually use the scanfilter for anything but WebBluetooth right now. We'll need to add it to macOS support I think.

psifertex, 2022-01-10:
FYI, based on https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158

I gave it a try by creating a signing certificate via key chain / create self-signed code signing cert

Then:

codesign -f -o runtime --timestamp -s "Signing Certificate Name" Bundle.app

And then it works fine!

metsis, 2022-03-17:
This seems to work again in macOS Monterey 12.3.

qdot, 2023-01-07:
This works fine on Monterey and is being distributed as part of programs now, closing.

closed 2023-01-07 by qdot