deviceplug/btleplug #231: Windows v0.29 will break UWP portion of library

qdot, 2021-12-19:
More a heads up than anything. windows v0.29 crate will break a bunch of windows stuff. A lot of types no longer implement Debug, and the new generator doesn't seem to bring in Send on some types, which is a problem when they're wrapped in IAsyncOperation. I've filed a couple of bugs over in windows-rs but I doubt there will be any movement until the new year. No rush to update the windows crate either, just something to be aware of.

qdot, 2022-01-17:
Ok, updating this for v0.30, which came out a few days ago. Most of the original breakage is gone thanks to the reimplementation of Debug, we're now down to a single struct that should be an enum (BluetoothAddressType). I'll file on this and we'll see what happens, but if we get stuck we can always make our own version.

qdot, 2022-01-17:
Filed at https://github.com/microsoft/windows-rs/issues/1437

hansmbakker, 2022-02-18:
Maybe this issue can be closed because of the fix in https://github.com/microsoft/windows-rs/issues/1437

qdot, 2022-07-02:
LAnded a while ago.

closed 2022-07-02 by qdot