deviceplug/btleplug #232: Handle encryption/bonding

Roughsketch, 2021-12-19:
I hope I'm not missing anything, but I'm unable to find how to do this at the very least.

I am trying to access a certain characteristic of a device (specifically Polar H10's "Control Point" characteristic), but am unable to since I just get back an insufficient encryption error. After lots of googling and looking at Polar's SDK, I am pretty sure I need to bond with the device to setup encryption before reading/writing to that characteristic. The problem is I don't see a way to do that in this library. I'm also not sure if there's a way to handle this externally since I think it requires the library to handle the encrypted communication.

Is there a workaround, plan to include/not include, or just some method I've missed to do this?

qdot, 2021-12-19:
Have you tried bonding the device with your OS first? That usually does the trick, we don't normally handle bonding inside the library.

Roughsketch, 2021-12-19:
I'm on Windows 10 so I went to bluetooth settings and paired with it, but it doesn't seem to change the output of the program.

[image]

Roughsketch, 2021-12-19:
And just to be clear in case I'm doing something else wrong, specifically what I'm doing is writing two bytes to the device and then getting back a longer response that I believe maps to insufficient encryption. I haven't had any hard confirmation of it besides the first byte matches that error code in several docs I've come across.

Basically:
Write `[2, 0]` (Start monitoring Ecg)
Read `[15, 5, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]` (15 matches Insufficient Encryption, not sure about rest).

qdot, 2021-12-20:
@Roughsketch What kinda bluetooth dongle/radio are you using? Broadcomm, CSR, Intel, RealTek, etc?

Roughsketch, 2021-12-21:
From device settings it looks like Intel. The device I'm using connects fine to Windows and shows up as paired, but I continue getting the same `[15, 5, 0, ...]` response. Is there any way to debug this on my end? 

Also the device itself is sort of a black box with an SDK I'm referencing, so if it sounds like this may not be BLE protocol related let me know. I don't want to steal your time on a specific side issue.

qdot, 2026-08-30:
Closing this out, mostly 'cause it's 5 years old, but also, bonding happens from the OS, not from our library. We have other devices that work with OS bonding fine, so this may be something specific to the device. 

Feel free to reopen or refile.

closed 2026-08-30 by qdot