deviceplug/btleplug #241: Stuff printed to terminal

flaviut, 2022-01-10:
**Describe the bug**

Sometimes I see things like `starting discovery failed DbusError(D-Bus error: Operation already in progress (org.bluez.Error.InProgress))` printed to the terminal.

This text isn't coming from my code, and I don't have logging enabled for btleplug.

**Expected behavior**

I expect nothing to be printed to the terminal unless logging is enabled using an environment variable.

**Actual behavior**

Things are printed to the terminal with logging disabled.

**Additional context**

Not really a big deal, but a bit annoying.

qwandor, 2022-01-10:
It turned out `bluez-async` had a `println!` rather than a `log::error!` so I've fixed that, but actually `btleplug` shouldn't have been calling `start_discovery_with_filter` without specifying an adapter so I'll fix that too.

flaviut, 2022-01-10:
That's a fun bug.

Thanks for working on this, I've found BLE to be a massive pain, but it's tolerable with this library :)

closed 2022-01-10 by qwandor