deviceplug/btleplug #243: Segmentation fault on Musl

mkb2091, 2022-01-17:
On linux x86_64 with musl as libc I get a seg fault when I create a manager.
`let manager = btleplug::platform::Manager::new().await;`

mkb2091, 2022-01-17:
there aren't any logs
```
warning: unused variable: `manager`
 --> src/main.rs:3:9
  |
3 |     let manager = btleplug::platform::Manager::new().await;
  |         ^^^^^^^ help: if this is intentional, prefix it with an underscore: `_manager`
  |
  = note: `#[warn(unused_variables)]` on by default

warning: `bttest` (bin "bttest") generated 1 warning
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `target/debug/bttest`
[1]    12549 segmentation fault (core dumped)  RUST_BACKTRACE=1 RUST_LIB_BACKTRACE=1 RUST_LOG=trace cargo run
```

I could upload the core file, though not sure where to upload as I don't think can just upload binary files to issues

As for using glibc on the same system, the system doesn't natively have glibc, but in a chroot with glibc it runs fine without seg faulting

qdot, 2026-08-30:
This is old enough now that I'm going to close out, but please reopen or refile if musl is still an issue.

closed 2026-08-30 by qdot