deviceplug/btleplug #244: Linux fails to discover services / characteristics

ryankurte, 2022-01-26:
**Describe the bug**
when connecting to a `YWK-J1` BLE SPO2 sensor `btleplug@0.9.1` inconsistently returns services in the `PeripheralProperties` object and `discover_services()` fails to discover any services or characteristics.

from messing with `gatttool` i _think_ this is because the device will only communicate when connected with an LE address type of `random` instead of `public`, `random` is returned in the properties option but doesn't seem to be used and i can't seem to find how to specify that here or via bluez-dbus.

**Expected behavior**

Discovery should list the characteristics available on the device, per `nrf connect` or `gatttool`:
```
➜  spo2 git:(main) sudo gatttool --device=DB:2B:7D:42:98:D0 -I -t random
[DB:2B:7D:42:98:D0][LE]> connect
Attempting to connect to DB:2B:7D:42:98:D0
Connection successful
[DB:2B:7D:42:98:D0][LE]> characteristics
handle: 0x0002, char properties: 0x0a, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0009, char properties: 0x02, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x02, char value handle: 0x000e, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x000f, char properties: 0x02, char value handle: 0x0010, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0011, char properties: 0x02, char value handle: 0x0012, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x0013, char properties: 0x02, char value handle: 0x0014, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x0015, char properties: 0x02, char value handle: 0x0016, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x0017, char properties: 0x02, char value handle: 0x0018, uuid: 00002a28-0000-1000-8000-00805f9b34fb
handle: 0x0019, char properties: 0x02, char value handle: 0x001a, uuid: 00002a50-0000-1000-8000-00805f9b34fb
handle: 0x001c, char properties: 0x08, char value handle: 0x001d, uuid: 00000002-0000-1000-8000-00805f9b34fb
handle: 0x001e, char properties: 0x10, char value handle: 0x001f, uuid: 00000003-0000-1000-8000-00805f9b34fb
handle: 0x0022, char properties: 0x0a, char value handle: 0x0023, uuid: 0000ff01-0000-1000-8000-00805f9b34fb
handle: 0x0025, char properties: 0x08, char value handle: 0x0026, uuid: f1080003-0451-4000-b000-000000000000
handle: 0x0027, char properties: 0x10, char value handle: 0x0028, uuid: f1080002-0451-4000-b000-000000000000
handle: 0x002b, char properties: 0x08, char value handle: 0x002c, uuid: f1020003-0451-4000-b000-000000000000
handle: 0x002d, char properties: 0x10, char value handle: 0x002e, uuid: f1020002-0451-4000-b000-000000000000
[DB:2B:7D:42:98:D0][LE]> quit
```

**Actual behavior**

Successful properties:
```
23:20:44 [INFO] Matching device!: PeripheralProperties { address: DB:2B:7D:42:98:D0, address_type: Some(Random), local_name: Some("J1"), tx_power_level: None, rssi: Some(-61), manufacturer_data: {60048: [208, 152, 66, 125, 43, 219]}, service_data: {}, services: [000090ea-0000-1000-8000-00805f9b34fb] }
```
Unsuccessful properties:
```
23:21:26 [INFO] Matching device!: PeripheralProperties { address: DB:2B:7D:42:98:D0, address_type: Some(Random), local_name: Some("J1"), tx_power_level: None, rssi: None, manufacturer_data: {60048: [208, 152, 66, 125, 43, 219]}, service_data: {}, services: [] }
```

regardless of properties, no services or characteristics are discovered.

**Additional context**
code is [here](https://github.com/ryankurte/rust-spo2).

- it's unclear to me whether this is an issue with btleplug (as `gattool` and `nrf connect` both function) or a non-standard BLE implementation that these tools compensate for.
- running this code with another known-good BLE device results in the expected list of characteristics.
- i tried to test this on macos to isolate the platform, but, (with terminal BLE permissions) scan doesn't seem to work at all
- i initially thought this was a regression so rolled back to `btleplug@0.8.1` which i had used previously with no change

ryankurte, 2022-01-27:
[bluer](https://github.com/bluez/bluer) does support [`ConnectDevice`](https://docs.rs/bluer/latest/bluer/struct.Adapter.html#method.connect_device) but it doesn't seem available in my version of bluez (`5.55-3.1` via debian repos).

there's also `l2_bdaddr_type` in `struct sockaddr_l2`, but i can't see how to use that API from here

qdot, 2026-08-30:
Closing now because this is 4+ years old, we'll need a new repro if it's still an issue, but enough has changed in the library that it's a refile situation at this point.

closed 2026-08-30 by qdot