deviceplug/btleplug #250: Exceptions from WinRT

hansmbakker, 2022-02-18:
**Describe the bug**
When running on Windows, I see lots of exceptions from Windows on the command line. Functionally it looks like it still works, but 

**Expected behavior**
Exceptions are not thrown

**Actual behavior**
Thousands of exceptions like the below are thrown
```
Exception thrown at 0x00007FF8C530478C (KernelBase.dll) in btleplug-experiment.exe: 0x0000071A: The remote procedure call was canceled, or if a call time-out was specified, the call timed out.
```
[image]

**Additional context**
Debug console logs:
```
-------------------------------------------------------------------
You may only use the C/C++ Extension for Visual Studio Code
with Visual Studio Code, Visual Studio or Visual Studio for Mac
software to help you develop and test your applications.
-------------------------------------------------------------------
Loaded '[C:\repos\btleplug-experiment\target\debug\btleplug-experiment.exe]()'. Symbols loaded.
Loaded '[C:\Windows\System32\ntdll.dll]()'. 
Loaded '[C:\Windows\System32\kernel32.dll]()'. 
Loaded '[C:\Windows\System32\KernelBase.dll]()'. 
Loaded '[C:\Windows\System32\oleaut32.dll]()'. 
Loaded '[C:\Windows\System32\msvcp_win.dll]()'. 
Loaded '[C:\Windows\System32\ucrtbase.dll]()'. 
Loaded '[C:\Windows\System32\combase.dll]()'. 
Loaded '[C:\Windows\System32\rpcrt4.dll]()'. 
Loaded '[C:\Windows\System32\vcruntime140.dll]()'. 
Loaded '[C:\Windows\System32\bcrypt.dll]()'. 
Loaded '[C:\Windows\System32\bcryptprimitives.dll]()'. 
Loaded '[C:\Windows\System32\ole32.dll]()'. 
Loaded '[C:\Windows\System32\gdi32.dll]()'. 
Loaded '[C:\Windows\System32\win32u.dll]()'. 
Loaded '[C:\Windows\System32\gdi32full.dll]()'. 
Loaded '[C:\Windows\System32\user32.dll]()'. 
Loaded '[C:\Windows\System32\imm32.dll]()'. 
Loaded '[C:\Windows\System32\kernel.appcore.dll]()'. 
Loaded '[C:\Windows\System32\msvcrt.dll]()'. 
Loaded '[C:\Windows\System32\clbcatq.dll]()'.
Loaded '[C:\Windows\System32\Windows.Devices.Radios.dll]()'. 
Loaded '[C:\Windows\System32\sechost.dll]()'. 
Loaded '[C:\Windows\System32\cfgmgr32.dll]()'. 
Unloaded '[C:\Windows\System32\cfgmgr32.dll]()'.
Loaded '[C:\Windows\System32\cfgmgr32.dll]()'. 
Unloaded '[C:\Windows\System32\cfgmgr32.dll]()'.
Loaded '[C:\Windows\System32\BthRadioMedia.dll]()'. 
Loaded '[C:\Windows\System32\devobj.dll]()'. 
Loaded '[C:\Windows\System32\WlanRadioManager.dll]()'. 
Loaded '[C:\Windows\System32\wlanapi.dll]()'. 
Loaded '[C:\Windows\System32\mobilenetworking.dll]()'. 
Loaded '[C:\Windows\System32\IPHLPAPI.DLL]()'. 
Loaded '[C:\Windows\System32\nsi.dll]()'. 
Loaded '[C:\Windows\System32\Windows.Devices.Bluetooth.dll]()'. 
Exception thrown at 0x00007FF8C530478C (KernelBase.dll) in btleplug-experiment.exe: WinRT originate error - 0x8000000B : 'The operation attempted to access data outside the valid range'.
Exception thrown at 0x00007FF8C530478C (KernelBase.dll) in btleplug-experiment.exe: 0x0000071A: The remote procedure call was canceled, or if a call time-out was specified, the call timed out.
Loaded '[C:\Windows\System32\SHCore.dll]()'. 
Loaded '[C:\Windows\System32\Microsoft.Bluetooth.Proxy.dll]()'. 
Loaded '[C:\Windows\System32\WinTypes.dll]()'. 
Exception thrown at 0x00007FF8C530478C (KernelBase.dll) in btleplug-experiment.exe: WinRT originate error - 0x8000000B : 'The operation attempted to access data outside the valid range'.
```

oletf, 2023-09-13:
I've been getting those too, 
Windows always has been being very slow at finding and connecting to devices, I think it might somehow be related to Advertising scan time windows...but not sure if WinRT allows to modify those.

oletf, 2023-09-18:
Here's more info if anyone wants to dig in deeper :

`Callstack`:
```
KernelBase.dll!RaiseException() (Unknown Source:0)
combase.dll!SendReport(HRESULT error, unsigned int cchMax, const wchar_t * message, unsigned short pSid, void * pExceptionObject, IUnknown *) Line 433 (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\onecore\com\combase\winrt\error\error.cpp:433)
combase.dll!RoOriginateError(HRESULT error, HSTRING__ * message) Line 590 (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\onecore\com\combase\winrt\error\error.cpp:590)
Windows.Devices.Bluetooth.dll!Windows::Foundation::Collections::Internal::Vector<class Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData *,struct Windows::Foundation::Collections::Internal::DefaultEqualityPredicate<class Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData *>,struct Windows::Foundation::Collections::Internal::DefaultLifetimeTraits<class Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData *>,struct Windows::Foundation::Collections::Internal::VectorOptions<class Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData *,0,1,0> >::GetAt(unsigned int,struct Windows::Devices::Bluetooth::Advertisement::IBluetoothLEManufacturerData * *) (Unknown Source:0)
rust_app.exe!windows::Windows::Foundation::Collections::IVector<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>::GetAt<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>(unsigned int self) Line 934 (c:\Users\user\.cargo\registry\src\index.crates.io-6f17d22bba15001f\windows-0.51.1\src\Windows\Foundation\Collections\mod.rs:934)
rust_app.exe!windows::Windows::Foundation::Collections::impl$119::next::closure$0<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>(windows::Windows::Foundation::Collections::impl$119::next::closure_env$0<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData> vector, windows::Windows::Foundation::Collections::IVector<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData> *) Line 1042 (c:\Users\user\.cargo\registry\src\index.crates.io-6f17d22bba15001f\windows-0.51.1\src\Windows\Foundation\Collections\mod.rs:1042)
rust_app.exe!enum2$<core::option::Option<ref$<windows::Windows::Foundation::Collections::IVector<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>>>>::and_then<ref$<windows::Windows::Foundation::Collections::IVector<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>>,windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData,windows::Windows::Foundation::Collections::impl$119::next::closure_env$0<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>>(enum2$<core::option::Option<ref$<windows::Windows::Foundation::Collections::IVector<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>>>> self, windows::Windows::Foundation::Collections::impl$119::next::closure_env$0<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData> f) Line 1413 (c:\Users\user\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1413)
rust_app.exe!windows::Windows::Foundation::Collections::impl$119::next<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>(windows::Windows::Foundation::Collections::VectorIterator<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData> * self) Line 1042 (c:\Users\user\.cargo\registry\src\index.crates.io-6f17d22bba15001f\windows-0.51.1\src\Windows\Foundation\Collections\mod.rs:1042)
rust_app.exe!core::iter::traits::iterator::Iterator::fold<windows::Windows::Foundation::Collections::VectorIterator<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>,tuple$<>,core::iter::adapters::map::map_fold::closure_env$0<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData,tuple$<u16,alloc::vec::Vec<u8,alloc::alloc::Global>>,tuple$<>,btleplug::winrtble::peripheral::impl$1::update_properties::closure_env$0,core::iter::traits::iterator::Iterator::for_each::call::closure_env$0<tuple$<u16,alloc::vec::Vec<u8,alloc::alloc::Global>>,hashbrown::map::impl$95::extend::closure_env$0<u16,alloc::vec::Vec<u8,alloc::alloc::Global>,std::collections::hash::map::RandomState,alloc::alloc::Global,core::iter::adapters::map::Map<windows::Windows::Foundation::Collections::VectorIterator<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>,btleplug::winrtble::peripheral::impl$1::update_properties::closure_env$0>>>>>(windows::Windows::Foundation::Collections::VectorIterator<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData> f, tuple$<>) Line 2480 (c:\Users\user\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\iter\traits\iterator.rs:2480)
rust_app.exe!core::iter::adapters::map::impl$2::fold<tuple$<u16,alloc::vec::Vec<u8,alloc::alloc::Global>>,windows::Windows::Foundation::Collections::VectorIterator<windows::Windows::Devices::Bluetooth::Advertisement::BluetoothLEManufacturerData>,btleplug::winrtble::peripheral::impl$1::update_properties::closure_env$0,tuple$<>,core::iter::traits::iterator::Iterator::for_each::call::closure_env$0<tuple$<u16,alloc::vec::Vec<u8,alloc: [... cut ...]
```

Seems to happen on calls of [`Peripheral::update_properties()`](https://github.com/deviceplug/btleplug/blob/13a638ae0f81cb47077aae7cefbee29495681b4b/src/winrtble/peripheral.rs#L134) with anything related to an `IVector` (WinRT zone I'm not comfortable with) gets iterated on any new advertising [Received](https://github.com/deviceplug/btleplug/blob/13a638ae0f81cb47077aae7cefbee29495681b4b/src/winrtble/ble/watcher.rs#L66)...

anyone knows if it's a problem at runtime or can be safely ignored ?